PubInv/general-purpose-alarm-device #8: Wired and Wireless Interface and Consequences.

ForrestErickson, 2022-06-11:
Many tangled issues In the section:

> How is the alarm signal received?
> There are many potential ways to communicate an API command to the GPAD. The obvious are:
> 
> The serial port
> SPI
> WiFi
> Bluetooth
> A wireless interface is probably most valuable for hobbyist and makers; and SPI-based protocol is probably most valuable for integration with a medical devices, such as [PolyVent Open Source Ventilator](https://www.pubinv.org/project/polyvent/). In general, we favor the use of JSON for a serial or wireless interface binding to the API, and a byte-level binding for an SPI connection.
> 
> Initial implementations should support a serial port, because this is the easiest to test and widely available on stand-alone Arduino devices. If the GPAD is integrated, it may forgo the serial port in favor of SPI.

Regarding, "and SPI-based protocol is probably most valuable for integration with a medical devices, such as... " comes assumptions that in addition the the four SPI signal wires is power and ground and if there is a button on the **general-alarm-device** the question arises of signaling back to the medical device the user's interaction (the button was pressed).  With this set of assumptions comes mounting hole pattern, with safe space left around mounting holes for hardware and providing end users drawings for mounting hole location versus the opening for display , lights, audio, and perhaps a reference design for a bezel.

For a NON integrated application where the  **general-alarm-device** is tethered on a cable, This SPI interface path will require the identification of a connector and pinout of the same. I know of no commercial products which put an SPI interface, power and ground on a cable (with the sole exception of a point of sale display, now obsolete, by Verifone, on an JR45 connector).  

With WiFi, and Bluetooth interface and the assumption of a complete enclosure comes the issue of identifying a power supply.  Is a WiFi / Bluetooth connection appropriate for a medical device (what happens with the RF connection is interrupted)?

If we really want a WiFi and Bluetooth interface the Arduino UNO is not the way to start development. The ARDUINO UNO WiFi REV2 may be the more appropriate device for proof of concept development (or an ESP32).  The use case for paring by Bluetooth will involve some button on the **general-alarm-device** .  
The connection by WiFi probably involves a WiFi manager to be set up by for example a smart phone to get the **general-alarm-device** on the hospital network and then the respirator to find the particular **general-alarm-device** to which to send the data and network security to prevent all the bad things bad players try to do to networks.

I (Lee Erickson) believe the decision on the requirement or lack there of for wireless interface needs to be made very soon and should reflect the very first proof of concept developed.

ForrestErickson, 2022-06-11:
For I2C interface in the maker community  there is the STEMMA interface. See: https://www.adafruit.com/category/1005

RobertLRead, 2022-06-11:
Good questions.

I assume that the GPAD should have a 12V jack.

There is not standard way to do SPI cabling, but we have used a shielded USB 4-wire cable up to 30cm to do I2C cabling. That is an acceptable solution, but the GPAD probably should just have a female 4-wire connector. I would just choose a JST connector that has a nice "snap" fit; this is always involves some tradeoffs, and I don't know a cookie-cutter way to do it. 

A WiFi or Bluetooth connection is good for hobbyist; bad for medical devices. Although there is some debate about wireless in ICUs, I want to avoid it; that is why I prefer SPI.  I would rather have an SPI connection on an UNO ahead of any other solution; however, if people want to start with a WiFi enabled microcontroller board I have no objection to that; but we need an SPI solution as a high priority.

The main need for a button on the alarm is to silence it. This is similar to the HUSH feature on kitchen smoke alarms; you push the button and the alarm is silenced for set time (let's say 2 minutes.) There is an academically recognized research area of Computer-Human Interaction as to how to do alarm management to prevent alarm fatigue for doctors. Basically, we can't solve this problem, but an MVP approach is just a button that silences for a fixed time. After the fixed time, if the alarm has not been sent to the lowest alarm level electronically, it starts to alarm again.  The beauty of this solution is that does NOT require anything to be communicated back to whatever device is generating the alarms; it does NOT need to know that the user silenced the alarm. However, adding that to the API could be a nice feature. 

WRT to setting up WiFi, Public Invention makes the VentMon device which is WiFi enabled. We implement a serial port protcol that lets you specify the SSID and password, which is stored in the EEPROM. This is done with the Arduino IDE serial monitor, typically.  This is completely adequate for engineers. However, you are correct, non-engineers probably need a phone app-based solution. However, since the WiFi use case is for hobbyist, I think we can go a long way with a similar "set the SSID and password with the serial port solution."

RobertLRead, 2022-06-11:
In case that is not perfectly clear: Rob thinks the MVP should not have WiFi.

ForrestErickson, 2022-06-12:
Regarding, "I assume that the GPAD should have a 12V jack."  We should begin to collect these requirements in probably more specialized documents and folders.   For example a surface mount 2.1mm barrel jack is: https://www.digikey.com/en/products/detail/tensility-international-corp/54-00164/10459298

I have been reluctant to mess with or add to the structure for fear not knowing enough of assumptions of how things have been organized in the past. Other than looking at for example MoonRat, is their a standard or template for organizing the project information?

RobertLRead, 2022-06-12:
I'm not aware of any such standard; certainly Public Invention does not
have one.

I personally would make the barrel jack size a story added to the project.

I consider the evolution of these "requirements" something that will happen
as we go along; for example, I personally don't know what the best (most
standard) specific format for a 12V barrel jack is off the top of my head.

LokiMetaSmith, 2022-06-22:
For wireless, I would use a System on a Chip, such as the ESP32 series. It's got good compatibility with existing arduino libraries, so the WiFi/Bluetooth feature would be a processor swap. Most of the peripherals should be compatible, as long as they accept 3.3v logic levels.

LokiMetaSmith, 2022-06-28:
Left as an option for another revision

closed 2022-06-28 by LokiMetaSmith