PubInv/general-purpose-alarm-device #10: Conventions for SPI Peripheral Device, LCD

ForrestErickson, 2022-06-12:
Their exist SPI LCD displays. The "general-alarm-device" is similar to such a device and may benefit end users if we follow those conventions. See for example the 16x2 display at DigiKey:

https://www.digikey.com/en/products/detail/orient-display/AMC1602AR-B-B6WTDW-SPI/12089324
The manufacturers data sheet is at: https://www.orientdisplay.com/wp-content/uploads/2019/10/AMC1602AR-B-B6WTDW-SPI-.pdf

The pin electrical interface is defined on page 8.
The pin mechanical interface is defined on page 9. Pin spacing is 0.1"

ForrestErickson, 2022-06-12:
A very inexpensive 16x2 LCD.

https://shop.robotistan.com/products/16x2-lcd-screen-green-on-black

LokiMetaSmith, 2022-06-28:
We decided to use the i2c interface for controlling the LCD directly, the SPI interface will be used to interface between the micro-controller, and the external interface will be spi