PubInv/general-purpose-alarm-device #11: Serial Transfer, SPI,

ForrestErickson, 2022-06-12:
**A very useful looking library for Serial Transfer.**

https://github.com/PowerBroker2/SerialTransfer

SerialTransfer is an easy to use Arduino library for transferring packetized data fast and reliably via Serial, I2C, and SPI interfaces. Interconnect your Arduinos today minus the headache with SerialTransfer!

LokiMetaSmith, 2022-06-28:
Closing for now, the library mentioned looks like it will work.

ForrestErickson, 2022-10-18:
Robert This is the  library I though might be worth looking at for sending serial data through the SPI interface.

ForrestErickson, 2022-11-22:
@RobertLRead 
@LokiMetaSmith 

Before closing I thought we should have some mutual review of the utility this library provides which I summarize in a quotation below.

Will the utility of this library be of use to PolyVent and GPAD?
What is the utility needed for our PolyVent and GPAD requirements?
Can the library help with the transfer of json files as I remember Robert was promoting?

> SerialTransfer is an easy to use Arduino library for transferring packetized data fast and reliably via Serial, I2C, and SPI interfaces. Interconnect your Arduinos today minus the headache with SerialTransfer!

[A mirror library is also available for Python!](https://github.com/PowerBroker2/pySerialTransfer)This library:
> 
> can be downloaded via the Arduino IDE's Libraries Manager (search "SerialTransfer.h")
> works with "software-serial", "software-I2C", and "software-SPI" libraries
> is non blocking
> uses packet delimiters
> uses consistent overhead byte stuffing
> uses CRC-8 (Polynomial 0x9B with lookup table)
> allows the use of dynamically sized packets (packets can have payload lengths anywhere from 1 to 254 bytes)
> supports user-specified callback functions
> can transfer bytes, ints, floats, structs, even large files like JPEGs and CSVs!!
>

ForrestErickson, 2022-11-22:
This library could be evaluated by using two GPADs.
As a start
For a "Controller" load the file:
https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/spi_tx_file/spi_tx_file.ino

For a "Peripheral" load the file:
https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/spi_rx_file/spi_rx_file.ino

There are other examples that may be of value.

RobertLRead, 2022-11-23:
I have looked at this, and I realized, we really don't want to use this for the serial port. The whole point of the serial port is to use a "plain text" message (which may be parsable in some, such as JSON.)

However, I think all of the packet construction (which is really what this library does) is VERY useful of SPI and I2C. So I still intend to use this.

Note I have (in a branch), create a python program that allows you to post things from the command line. This is sort of a useful point for some people who want to use Python. It needs to be done.

ForrestErickson, 2022-11-23:
@RobertLRead 
Regarding, "Note I have (in a branch), create a python program that allows you to post things from the command line. This is sort of a useful point for some people who want to use Python. It needs to be done."

It is possible from the Windows command window to send commands such as:
echo a4TEST > \\.\COM15

I have tried this and the GPAD is receiving something but not parcing it correctly. The COM terminal is sending a CR LF termination.

ForrestErickson, 2022-11-23:
This is closer. The GPAD_API V0.10 responded to this command.
I also displayed the COM port setup for a reference.

[image]

ForrestErickson, 2022-12-03:
@RobertLRead reports the following in an email of 20221202:

>I have used the "Serial Transfer" library to transfer strings via SPI over our cable between the two boards with good connectors; this is good progress.
Sadly, however, the owner of that repo communicated to me (via an issue) that he no longer supports SPI, and I had to modify the files by hand to make them work. This produces a strategic quandary: use a library that has to be modified by hand, or write our own, largely duplicating the functionality that is in that library?
I will think about this.

ForrestErickson, 2022-12-03:
@RobertLRead 
Could you copy or link to the correspondence with the author of the library here in this issue please?

LokiMetaSmith, 2023-01-03:
Robert had issues compiling on the ESP32 with the library and so we will not be using this library.

closed 2023-01-03 by LokiMetaSmith