PubInv/general-purpose-alarm-device #9: Requirement for Text Display,  References to Liquidcyrstal library. Identifying Hardware Interface

ForrestErickson, 2022-06-12:
**This issue is at root about choosing interface pins on and LCD so as to not conflict with the SPI Peripheral pin choices.**

From the Project Read me, 

> The Abstractly, we can imagine a system that supports five levels of alarm severity. Additionally, an alarm condition normally has a short message associated with it, such as "hose disconnect", which should potentially be displayed on a small screen.

A logical way to support this requirement is to use an LCD character displays in Arduino library are found at: 
https://www.arduino.cc/reference/en/libraries/liquidcrystal/

The "HelloWorld" example has pin connections for the Arduino UNO as

` The circuit:
 * LCD RS pin to digital pin 12
 * LCD Enable pin to digital pin 11
 * LCD D4 pin to digital pin 5
 * LCD D5 pin to digital pin 4
 * LCD D6 pin to digital pin 3
 * LCD D7 pin to digital pin 2
 * LCD R/W pin to ground
 * LCD VSS pin to ground
 * LCD VCC pin to 5V
 * 10K resistor:
 * ends to +5V and ground
 * wiper to LCD VO pin (pin 3)`

However, The SPI slave interface uses some of the same pins.
`/* Hardware Notes Peripheral
   SPI Line Pin in Arduino, IO setup
  MOSI 11 or ICSP-4  Input
  MISO 12 or ICSP-1 Output
  SCK 13 or ICSP-3  Input
  SS 10 Input
*/`

So for either the LCD RS and Enable pins  or the SPI MOSI and MISO pins we have to modify software.  

The SPI Peripheral is time sensitive code Lee has implemented at: https://github.com/ForrestErickson/SPI_Controller_And_Peripheral/blob/main/SPI_PERIPHERAL/SPI_PERIPHERAL.ino
It uses an interrupt for capture of the data in the serial shift registers of the UNO.   Probably not relocatable.

The LCD pins for RS and ENABLE the UNO look to be settable in the library . From the HelloWorld example code

`// initialize the library by associating any needed LCD interface pin
// with the Arduino pin number it is connected to
const int rs = 12, en = 11, d4 = 5, d5 = 4, d6 = 3, d7 = 2;
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);
`

Looks like we could try for example, 
`// initialize the library by associating any needed LCD interface pin
// with the Arduino pin number it is connected to
const int rs = 10, en = 9, d4 = 5, d5 = 4, d6 = 3, d7 = 2;
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);
`

RobertLRead, 2022-06-12:
A more reasonable approach is to find a display that uses I2C or SPI, so that it does not use so many pins.

ForrestErickson, 2022-06-20:
I spent some time at DigiKey and then Aliexpress. There is quite the price difference with a 20x4 display including the adapter board for IIC interface at only US $3.66 with Shipping: US $4.48 .  Lead time is August 8 - June 20 or about 6-7 weeks.
https://www.aliexpress.com/item/2255800949200886.html

At quantity of 5 the shipping only increases to Shipping: US $10.38. This works out to (5*$3.66+1038)/5= $5.74 each at QT=5.

And parts available anywhere Aliexpress ships.

However, their are no mechanical nor electrical specifications at the link. So that needs research.
The I2C interface board shows a PCF8574.  There is an ARDUINO library for interfacing to LCD with the PCF8574 at: https://www.arduino.cc/reference/en/libraries/liquidcrystal_pcf8574/

Lee has what appears to be Identical hardware QT of 1 in his lab from which physical measurements can be taken to compare against any drawings if found.

ForrestErickson, 2022-06-20:
The Aliexpress represents the low cost if you are willing to do quantity batches way and schedule.

In stock at Amazon now is: 
https://www.amazon.com/GeeekPi-Interface-Adapter-Backlight-Raspberry/dp/B07QLRD3TM/ref=sr_1_2
It is about twice the price at:  $9.99

It looks to be the identical part.

In our yet to be invented parts master the part number for this display should capture both vendors.

RobertLRead, 2022-06-20:
Until we go into high production, I consider the cost largely irrelevant.
I'll be happy if we make good progress on the prototype; that is a far
greater risk. If we choose a part that is too expensive, it will be
relatively easy for us to swap it out for a cheaper part later; that is an
advantage of a fully open design and prototyping approach.

LokiMetaSmith, 2022-06-28:
The schematic uses an i2c interface to the microcontroller, should have an Arduino library and can be expanded through the external i2c interface connector.