PubInv/general-purpose-alarm-device #13: Cable Connection for I2C Device, COTS

ForrestErickson, 2022-06-21:
A commercial off the shelf cable for I2C is the one from Adafruit.

STEMMA QT / Qwiic JST SH 4-pin Cable - 100mm Long
Candidate cable: https://www.adafruit.com/product/4210

To Do for PCB:
Add a jack compatible with this on the GPAD PCB so that we can daisy chain to other devices.

RobertLRead, 2022-06-21:
This is good but that connector doesn't snap in as much as I would like it
to. It is quite small and can come out with pressure or vibration. However,
the same cable with a more solid connector should be acceptable.

ForrestErickson, 2022-06-22:
An RJ12 is 6P6C and would have conductors enough for signals and redundant power and ground.   

We would need PROMINENT labeling for its purpose so that it would not be misused. 
We would need to PROMINENT identify that a data cable (Wired pin 1 to pin 1) must be used and not a telephone cable (Wired pin 1 to pin 6).
Examples from amazon: https://www.amazon.com/s?k=rj12+6p6c+data+cable&gclid=CjwKCAjw-8qVBhANEiwAfjXLrmktj8ARJSNiqknd2nZTo-efkqLtqwy3ivh5r2o4UW3t-lKrgdihGxoCCf4QAvD_BwE&hvadid=557591226617&hvdev=c&hvlocphy=9013385&hvnetw=g&hvqmt=e&hvrand=13948863055008248823&hvtargid=kwd-427348137102&hydadcr=18911_13357683&tag=googhydr-20&ref=pd_sl_7lmopi145_e

LokiMetaSmith, 2022-06-22:
RJ-12 have connector stability issues and drastically increase bulk. 

JST-SM have options for arm locking mechanisms which would make them resistant to pulling out. 

http://www.jst.co.uk/productSeries.php?pid=94&cat=117

LokiMetaSmith, 2022-06-28:
Decided to keep with a JST style connector, but the JST-SM is a wire to wire only connector

Alternatives are. 
JST-VH
JST-XA
JST-PA

https://en.wikipedia.org/wiki/JST_connector

ForrestErickson, 2022-06-30:
The JST-PA series is 2mm which will be compact similar to the 2.54 (0.100 KK series)
at JLCPCB a for conductor part is quite expensive and not in stock..

For Vertical pins:
JLCPCB has B04B-XASK-1(LF)(SN) in stock. JLCPCB part number:  C264994  Price: $0.0735
At Digi-key is a 2.54mm part 455-1994-ND
In stock and price at QT 1 = $0.30
https://www.jst-mfg.com/product/pdf/eng/eXA1.pdf

Mating plug appears to be XARP-04V, Digi-Key Part Number: 455-4152-ND at $0.19000,
JLCPCB had no stock on the plug.

ForrestErickson, 2022-08-02:
Should we be identifying some cable assemblies with the mating connector to order or order the materials necessary to build same?

Having a cable would allow us to check mechanical fit with the enclosure.
Is their a clearance requirement impacting the enclosure?

Order cables in quantity of 15.
Ship to Lawrence. 
Lee to identify something to order.

ForrestErickson, 2022-08-02:
For the header S04B-XASK-1(LF)(SN), the mating part shell is: https://www.digikey.com/en/products/detail/jst-sales-america-inc/XARP-04V/9951661  455-4152-ND

The 	Socket Contact Tin 22-26 AWG Crimp is: 455-2050-1-ND

LokiMetaSmith, 2022-08-04:
This looks resolved, 

JST-XS and JST-XH share the same crimp shape and the crimp tools are interchangeable. 

https://www.amazon.com/XH2-54mm-XH3-96mm-Terminals-SN-01BM-%EF%BC%880-08-0-5mm2%EF%BC%89/dp/B00YGLKBSK/

ForrestErickson, 2022-08-04:
I must admit to fatigue back on 20220802.  The part for which the board is designed is in stock. 
IIRC, I am frustrated that I cannot find the mating connector parts or prebuild cables. 
This feature (The I2C output) is a wish not a must for V1 and I think we should build V1 assemblies with the design as is and keep trying to find the mating part as time allows.

This issue is NOT ready to close.

LokiMetaSmith, 2022-08-09:
Agreed, we will leave this open until we've tested the connectors out on V1 when those are ordered.

LokiMetaSmith, 2022-08-16:
The digikey order will include compatible plug and male pin connectors to test this out. The purpose of this cable is to jumper between GPAD devices and to allow for cutting in half to provide pig tail connectors for future expansion. 
plug shell, with crimpped pins will be ordered and applied to at minimum two cable assemblies.

LokiMetaSmith, 2022-10-04:
Plan on testing mechanical fit by October 5th

ForrestErickson, 2022-10-25:
We have built Version 1 with the JST-XS and JST-XH share the same crimp shape and the crimp tools are interchangeable.

closed 2022-10-25 by ForrestErickson