PubInv/general-purpose-alarm-device #14: Complete schematic symbols and foot prints for GPAD proof of concept

RobertLRead, 2022-06-22:


LokiMetaSmith, 2022-07-27:
We've reviewed the PCB placement with respect to components that affect the enclosure, there are some changes to the i2c connector is a place holder at the moment. 

Change J102 and J103 from a connector to a header
Float the display off the board
Change the test points so they show the value, and see if shrinking the size of the Ref. Designation helps
Layout footprint for 3 terminal 7805 regulator

ForrestErickson, 2022-07-28:
RE: Change J102 and J103 from a connector to a header
Done:
[image]

Re: Float the display off the board. The spacer thickness is .125" or 3.175mm
Done:
[image]

Re: Layout footprint for 3 terminal 7805 regulator
Found horizontal foot print.
[image]

ForrestErickson, 2022-07-28:
RE: Change the test points so they show the value, and see if shrinking the size of the Ref. Designation helps
Done.
[image]

All the issues found in the review of 20220727 cleared.

closed 2022-07-28 by ForrestErickson