PubInv/general-purpose-alarm-device #17: Enclosure Requirements, Proof of Concept, Prototype, Product

ForrestErickson, 2022-06-22:
Regarding, 
`Thoughts On An Initial Enclosure
An initial device about the size of a bar of soap that could be easily mounted with zip ties onto some other device of external frame would be very useful. The utility of additional miniaturization would be marginal. The Arduino UNO is about this size to begin with. I normally begin projects with an UNO. A soldered UNO Shield would be a quick way to get an initial prototype working. A mechanical enclosure that project LEDS slighly and let a buzzer or loudspeaker operate loudly (through a grate or grid) would probably be adequate.`

For the proof of concept Lee assumes no enclosure. It is a spider web of modules and cables and solderless breadboard.

For a prototype there need not be an enclosure because the "customers" for the very first prototypes are the developers of the GPAD firmware and the developers of the ventilator (Other product developers). However the PCB should be designed for an enclosure to follow. 
An enclosure (3D plastic printing?) must have openings for the LCD, for power and SPI signal connection, power connection, indicator lights (5?). Acoustic transducer, Daisy chain SPI (I2C) signal and power connections.  Design for a custom button with an off the shelf spring will likely be required. as are PCB into enclosure mounting method definition and enclosure mounting to outside world mounting method definition.

For a final product, is their (should their be) a requirement for washdown?  A requirement for high temperature sterilization?
A quick look at the IP designation wikipedia page does NOT show a medical wash down example. See: https://en.wikipedia.org/wiki/IP_Code

Google did not help me find a specification either.

Lee

LokiMetaSmith, 2022-06-28:
We've partly addressed this and added some documentation to the ReadMe, in the Product Description section.

LokiMetaSmith, 2022-07-05:
I found this project for designing and 3D printing parametrically defined project boxes. 
https://hackaday.com/2018/03/02/printed-it-custom-enclosure-generator/
https://www.thingiverse.com/thing:1264391/files

It looks comprehensive and perfect for our application.

LokiMetaSmith, 2022-07-19:
It appears the enclosure I've found is sufficient after discussion and review.

closed 2022-07-19 by LokiMetaSmith