PubInv/general-purpose-alarm-device #19: Research: The OSVI Alarm System...

RobertLRead, 2022-06-23:
I recently stumbled on this ventilator [alarm system](https://gitlab.com/ovsi-org/ovsi-ventilator/-/tree/master/Design/Electronics%20%26%20Control/Alarm%20system) which is similar to what we are trying to do with GPAD.

LokiMetaSmith, 2022-06-28:
We need to review and incorporate parts of this project into ours. 

We can do a project review at a later date.

ForrestErickson, 2022-10-06:
I found the link to the abstract of one of the referenced documents at:
https://www.iso.org/standard/72069.html and the full text for free at:

**https://www.iso.org/obp/ui#iso:std:iso:80601:-2-12:ed-2:v1:en**

ForrestErickson, 2022-10-06:
Robert,

Upon reading the project at gitlab "OVSI Open Ventilator System InitiativeOVSI Open Ventilator System Initiative"
I am compelled to state that we will need to case by case consider the implications that the GPAD is a feathered peripheral which if the communication to the main have to have additional default behaviors than implied in the ISO:80601 which, as I read it, may not assume a tethered display and user interface.

Should the GPAD Peripheral become disconnected from the Controller it must inform the user of this state. The urgency of expression of this disconnection should be **TBD.**   Perhaps there should be a time element similar to the 120s (two minute) limit on muting an audio alarm.

Lee believes this should be a feature and request issue over and above the features implied or required by ISO:80601.

ForrestErickson, 2022-10-06:
The ISO:80601 references another document, IEC 62366-1:2015(en)
Medical devices — Part 1: Application of usability engineering to medical devices

Which Lee found (but have not fully read) here:

https://www.iso.org/obp/ui#iso:std:iec:62366:-1:ed-1:v1:en:sec:5.6,5.8

RobertLRead, 2022-10-27:
I have read all that is available in the ISO standard (without paying for it.) I personally don't think we need to worry about that. We are very far away from FDA clearance, so in the short term, I see little advantage in trying to comply with such a standard.

The OSVI work has one feature that I think is worth coding: The use a multiple "beeps" instead of a steady tone for the alarm condition. I think this is very nice feature for us to have, and I will create a separate issue for it.  I recommend that we close this in our next meeting.

RobertLRead, 2022-11-01:
This issues functionality is being pushed into other issues.

closed 2022-11-01 by RobertLRead