PubInv/general-purpose-alarm-device #20: Porting kicad 5 to kicad 6

LokiMetaSmith, 2022-06-28:
We need to port the project over to kicad v6 when we have a good stopping point, probably the schematic capture

ForrestErickson, 2022-07-05:
On 20220705 Lawrence (LokiMetaSmith) and Forrest Lee Erickson reviewed the schematic. Lee made very ugly notes for follow up.
[image]

LokiMetaSmith, 2022-07-12:
Lee will push the update.

ForrestErickson, 2022-07-12:
Lee made a tag on the "LastKiCad5Commit in the repository at:
https://github.com/PubInv/general-alarm-device/releases/tag/LastKiCad5Commit

ForrestErickson, 2022-07-12:
Notes on opening with KiCad6.
[image]

ForrestErickson, 2022-07-12:
I have noticed than when I made the KiCad6 commit that only one file was changed the PCB. 

It looks like schematic files are now .kicad_sch files. I see in my folder that the .sch files still have the 12:22 PM time stamp but the new .kicad_sch files have the 7:18 PM time stamp.

[image]

Looks like I need to add the "*.kicad_sch" to the tracked files.
And the .kicad_pro file too.

**$ git push
Enumerating objects: 15, done.
Counting objects: 100% (15/15), done.
Delta compression using up to 4 threads
Compressing objects: 100% (12/12), done.
Writing objects: 100% (12/12), 119.51 KiB | 5.98 MiB/s, done.
Total 12 (delta 5), reused 0 (delta 0), pack-reused 0
remote: Resolving deltas: 100% (5/5), completed with 1 local object.
To https://github.com/PubInv/general-alarm-device
   30100a7..2c6b165  main -> main

Admin@Melies MINGW64 /c/Users/Public/Downloads/Arduino/PubInv/general-alarm-device (main)
$ git log
commit 2c6b1651a6e4d4274074dd90a7035e0f9c54fe22 (HEAD -> main, origin/main, origin/HEAD, AddNewFileTypesKiCade6)
Author: Forrest Erickson <forresterickson@comcast.net>
Date:   Tue Jul 12 19:52:51 2022 -0400

    Add some new file types for KiCade6.
**

ForrestErickson, 2022-07-13:
FYI: I have left the following question on the KiCad Facebook group.

`I have migrated a complex design from KiCad 5 to 6.
Several schematic symbols are messed up. They are from a project library I think.
When I open the symbol editor and try to open the symbol affected I get this message in the dialog telling me to Use Manage Symbol Libraries to migrate to current format.
Anyone know where oh where is a "Manage Symbol Libraries" to be found?`

ForrestErickson, 2022-07-13:
Ok.
Looks like I found it.
[image]

Will have to investigate tomorrow.

ForrestErickson, 2022-07-15:
Lawrence,
Regarding, "I believe this is mostly complete at this point."  I noticed you had committed some changes. I will look soon.

FYI, I have been practicing some KiCad 6 since yesterday.  I wanted to make an UNO shield for a power interrupt test fixture for some time and so did. I included the JR12 connector with SPI connections compatible with GPAD too.
It is here: https://github.com/ForrestErickson/PowerInterruptUNO

Regarding, Robert's "The PolyVent side does not have a well developed fault detection, and if it
did, it would not be easily to mechanically induce all of the faults. We
definitely need a test suite that creates artificial faults. In fact, I
would say we should develop a test suite that has nothing to do with
PolyVent. If needed I can write code for that pretty quickly."

Code written for an UNO with this shield would be a simulated PolyVent or what ever we want to call it.
[image]

ForrestErickson, 2022-07-18:
On about Friday 20220715 Lee pulled the updates from the github origin repository.
On 20220718 Lee again looked at the schematic symbol libraries and found they were still not editable.  Got brave and made a branch on which to try the Manage Symbol Libraries, and the Migrate Libraries button.
[image]

The dialog closed uneventfully.
The library no longer appeared in the list of libraries. Perhaps closing and reopening will work. Nope. 

Double click on the GPAD_ACH.lib in the project and get this error:
[image]

Closing KiCad 6 to try again.

ForrestErickson, 2022-07-18:
Reopened KiCad6 Opened the symbol editor. Again selected Preferences, Migrate Libraries and again selected the GPAD_ACH_LIB to migrate. Pressed Yes.  The dialog closed. This time after about 30 to 60 seconds the Symbol Editor dialog refreshed and the library is there!!!
I can open, for example, the buzzer.  
[image]

Hurray!

I have pushed the project with the migrated library  to Github.

While there is more work to do on the design Lee believes the migration to KiCad 6 is done.

ForrestErickson, 2022-07-18:
Additional Notes. I found the LDC module was still referencing the rescued library from KiCad5.  When I brought in the symbol from the GPAD_SCH_LIB the DRC then through an error on pins 7,8,9 and 10 which are grounded.

KiCad5 Library part with pins 7, 8, 9 and 10 set as input.
[image]
Default library part with pins 7, 8, 9 and 10 set as Bidirectional.
[image]

I have changed the project library to be like the KiCad 5 version with pins 7, 8, 9 and 10 set as input.

I updated the symbol from the library and the DRC errors associated with U302 are cleared.

**NOTE; that this exercise reflects a possible bad thing that could happen. If the LCD were to be read from then these pins would be trying to drive a signal on GND.**

ForrestErickson, 2022-07-18:
Lee has connected everything approximately where Lawrence placed components. HOWEVER, Their is probably a mix of parts on metric grid and imperial grid. UGH!!!  Wish I had realized I was probably fighting grids earlier.

Regardless **This layout is probably a throw away anyway**. Lee only has about five hours invested and was learning KiCad 6 work flow anyway.

**Issues:**

1. Lee did not move the LEDs as per our discussion of 20220712. That is why likely throw away.
2. There is poor clearance for LCD mounting hardware and the UNO headers.
3. The reset switch S101 probably clears the UNO shield but is very close.
4. Their is a larger hole in the bottom layer ground than could probably be achieved with a restart. Regardless this design has enough ground that Lee would guess it would pass FCC even if we tried to make it fail by making every IO we have connected run at the same time.
5. No electrical DRC errors. However there are numerous errors with silk screen outlines to the UNO and the board edge.

DRC Report except silk screen errors
[image]

The layout top and bottom
[image]
The layout bottom (mostly ground only)
[image]

The 3D from component side
[image]
The 3D from LCD side.
[image]

ForrestErickson, 2022-07-18:
Changes pushed at 20220718 18:26 EDT.

LokiMetaSmith, 2022-07-19:
Library changes have been made, and no current lingering issues from migrating from kicad 5 to 6

closed 2022-07-19 by LokiMetaSmith