PubInv/general-purpose-alarm-device #21: Add some example strings for alarms coming from the PolyVent Ventilator

RobertLRead, 2022-07-05:


RobertLRead, 2022-07-05:
Here are some example calls:

`
int Alarm80(PANIC,"Hose Disconnect. No pressure in the airway.");
int Alarm80(PROBLEM,"PVA: asynchrony at 50% for 5 minutes.");
int Alarm32(WARNING,"FiO2 levels out of spec.");
int Alarm80(WARNING,"FiO2 levels out of spec. FiO2 low by more than 5% for 37% of last 5 minutes.");
int Alarm32(CRITICAL,"Backup battery indicates low battery life.");
`

RobertLRead, 2022-07-05:
@ForrestErickson I did this is quickly, does this answer your question?

ForrestErickson, 2022-07-08:
Regarding, " I did this is quickly, does this answer your question?"
Not at all what I expected. Which is why we ask these questions.

I was expecting something documenting the data that when through the byte interface so more like

`LCD 0,0 Hello World<CR>`
Where "LCD" indicated the destination. "0x0" was the cursor location, "Hello World" was data and "<CR>" was a line termination such as a Carriage Return or a Line Feed for both.

Perhaps also some special functions such as:
`LCD RESET<CR>` 
to clear the display.

`LCD CLEAR_ROW 1<CR>`  
to clear the second line.

And 
`LCD 1,0 Second Line Text `
to put text on the second line.

And 
`LCD CLEAR_ROW 1<CR>`  
`LCD 1,0 ## <CR>`
to put two hash marks in the center of the second line.

For lights 0 through 4 each of which can have 1-255 levels, something like
`LIGHT 255 255 128 0 0<CR>`  
to have lights 0 and 1 fill lit and 2 half lit and 3 and 4 off.

For the audio alert something like
`TONE 440 <CR>`   for a steady tone at 440 Hz
`TONE 440 1000 200<CR>`   for a tone at 440 Hz on for one second and off for 200 mS.
`TONE 0<CR>`   for a tone off.

RobertLRead, 2022-07-08:
Well, the user of the C Library for the GPAD will not be making calls specifically to set TONE, LIGHT, and LCD as you have suggested. I believe the user will call 

Alarm80(PANIC,"Hose Disconnect. No pressure in the airway.");

That will then get translated into some byte-level communication which will be sent by SPI. Then, on the GPAD side, there will be code very much as you have written. That is, it is the job of the GPAD's resident code to determine how the get the LCD to display "Hose Disconnect. No pressure in the airway.". Likewise, it will be that code's job to figure how how to turn on the light, and how brightly, or flashing at what rate, to represent PANIC. Similarly, it must determine what the sound for PANIC will be.

This allows different devices implmenting the GPAD interface, or evolving versions over time, to change how they treat the very abstract command: Alarm80(PANIC,"Hose Disconnect. No pressure in the airway."); in different way.

Is that clear?

ForrestErickson, 2022-08-14:
Regarding, "Well, the user of the C Library for the GPAD will not be making calls specifically to set TONE, LIGHT, and LCD as you have suggested. I believe the user will call"

I think this is hat I was calling the GPAD_LIB in the email I wrote  a bit after midnight, morning of 20220814 email.

I would think we instantiate a GPAD device with /CS pin as per the Controller wiring.
The device would get a name such as myGPAD or firstGPAD.
Their would be a method to set the /CS pin number.
Their might be a firstGPAD.begin() which is called in setup. It might return success or failure of the connection.
Their would be a method to ?print? to the GPAD such as firstGPAD.alarm(PANIC, "Hose Disconnect. No Pressure in the airway.");
Other commands
firstGPAD(NORMAL, "Normal.");
firstGPAD(WARNING, "Check battery life");

If a second GPAD is in the system it might be
secondGPAD(CRITICAL, ""FiO2 levels out of spec.");

RobertLRead, 2022-08-16:
I think this will be handled by creating two different objects.

For example, gpad1 and gapd2.
Then we just exeucte:
gpad1.alarm(NORMAL,"Normal").

So I suggest this issue can now be closed.

ForrestErickson, 2022-10-25:
This has served the purpose of promoting discussion and efforts are now around the actual GPAD_API code.

closed 2022-10-25 by ForrestErickson