PubInv/general-purpose-alarm-device #22: Command Protocol, Delimiters,

ForrestErickson, 2022-07-05:
The command API must have some delimiters to indicate end of the command.
Possibly to separate multiple commands in one message.
Commands to identify the action of the GPAD sub section such as TEXT, SOUND, LIGHT, VIBRATION, OTHER
Augments to modify the action such as what text to display. What sound to make and how long, What light to light and how bright and how long and flash. What tone to make and for how long and how loud.

**Ideas to examine for developing.** 

- The commands for LCD display and the library for same.
- Standard Commands for Programable Instruments (IEE488) might offer ideas too. See:  https://en.wikipedia.org/wiki/Standard_Commands_for_Programmable_Instruments 
 (Note Command start with colon ":", multiple commands delimited by ";", command arguments by space" " )
- The libraries for I2C or SPI addressable LED strings should be examined for insight.
- The read and write to an SPI device such as a EEPROM might suggest ideas.

ForrestErickson, 2022-07-05:
LCD programing example for what looks to be exactly the same hardware (I2C to nibble) and 20x4 character display with code at:

https://www.instructables.com/Connect-LCDs-With-NodeMCU-and-Arduino/

ForrestErickson, 2022-10-25:
Robert is skeptical of the use case. Will read.

RobertLRead, 2022-10-27:
I've read the SCPI documentation above. I think that standard is far removed from JSON which is now much more common for most of our users. I recommend we not use SCPI until such time as a user requests. Although certainly LabView compatibility is a nice-to-have features, I feel it would be aiming too far ahead of our current state to go there. I doubt we would attract a single user in the next year from LabView unless we are very successful with other more hobby-oriented audiences. I recommend that we close this issue in our next meeting.

ForrestErickson, 2022-10-28:
I understand.

By the way there is a feature of the SCPI standard commands where the device returns the manufacturer and model and perhaps software version number which I think is interesting.   It is the "IDN" command.

There is also a RESet command.

I implemented something like this on my OCTOUNO firmware found here:
https://github.com/ForrestErickson/OctoUNO/blob/main/OctoUNO.ino

HOWVER, I do not have anything like a full SCPI parser in this project.

ForrestErickson, 2022-11-01:
We have discussed and can close as no further action.

closed 2022-11-01 by ForrestErickson