PubInv/general-purpose-alarm-device #23: Update schematic per Review of 20220705

ForrestErickson, 2022-07-05:
1. Limit current from / to VBus
2. Limit current from / to ControllerVcc 
3. Limit current at barrel input
4. Move C101 to cathode D101 (Will not fix, C101 is to limit rise time on differential mode on power cable for EMC)
5. Add resistive divider to set a fixed contrast on LCD at U302 V0 pin.
6. Add fuse near J501 (USB connector) on power.  (Will not fix, current from USB supplies are limited already to 100mA)
7. Add schottky diode foot print across J102, J103.
8. Should ControllerVcc connect to Input pin 3 of U101?  This is the same as asking do we want the poly vent to supply 12V to the GPAD?
9. Add a full Arduino UNO footprint so that rather than populating U101 and U501 we use the GPAD as a bottom mounted shield for an UNO.  NOTE: Allow for holding on board U101 and U501 in separate reset? if using and UNO?
10. Move I2C pull up resistors to top schematic out of LCD sub page.
11. Add cap to "debounce" S601.
12. Add series resistor to BZ601 to limit current.
13. Add parallel connector to BQ601 for an off board transductor.
14. Check and change schematic symbol for D105, for schottky.

ForrestErickson, 2022-07-09:
20220709.  While adding the UNO schematic symbol and PCB footprint Lee realized that the 3.3V pin has nothing in our design with which to connect. 

In the UNO Rev 3 schematic there is a 3.3 V LDO and an op amp circuit switching from Vin to USBVCC through a FET to generate this 3.3V supply. 
[image]

I am going to leave these components out and make the 3.3V pin a No Connect in the GPAD unless someone wants to make a compelling use case.  Adding the UNO foot print is already mission creep. 

A NANO foot print might actually be a better idea.

ForrestErickson, 2022-07-09:
20220709 11:50 or so. Lee has pushed the changes into the repository.

Several files in the local folders not yet under control. Many are just vendor diagrams for which Lee may want to add some sub folders and rename. However there are some such as, "GeneralPurposeAlarmDevicePCB.pretty/BarrelJack_CLIFF_FC681465S_SMT_Horizontal.kicad_mod"
that perhaps are needed but Lee does not fully understand.
See:
[image]

ForrestErickson, 2022-07-09:
Schematic main page screen shot showing with in the sheet the UNO symbol and to the right the NANO symbol:
[image]

PCB with foot prints pulled in:
[image]
And the 3D render such as it is:
[image]

ForrestErickson, 2022-07-09:
All schematic components have foot prints. It might be a good idea for a review of them with fresh eyes.
[image]

RobertLRead, 2022-07-09:
> 

I think people do generally use the Nano if they are integrating it onto a board; but I don't have a lot of experience with this.

ForrestErickson, 2022-07-09:
End of day 20220709 and I still need to get a micro USB part for which I can find, stock, a schematic symbol, a PCB footprint and 3D model.
Currently working towards: https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C14996 or https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C132563 which are NOT foot print compatible (At least this late in the day I do not think they are.)

Same for a buzzer of some sort.  Currently working towards: https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C96256

ForrestErickson, 2022-07-10:
From the KiCAD Facebook group, Lee learned about componentsearchengine.com
Found a schematic symbol and PCB symbol for QMB-09B-03.  https://componentsearchengine.com/part-preview/QMB-09B-03/Jiangsu%20Huaneng%20Elec?type=3d

However there was still no 3D model even though one was pictured.

[image]

My parts management and library management is become chaotic.

ForrestErickson, 2022-07-10:
Reading the license for the part I down loaded makes me not sure we can incorporate them and "redistribute" them.  NOT SURE.

[image]

LokiMetaSmith, 2022-07-11:
With regard to the part license, I would include it as we are not distributing it as part of a separate library component and it's tied directly to the General Alarm Device.

RobertLRead, 2022-07-12:
NOTE: We have decided to move all the cable connections to the "bottom edged" of the case. At present, Lee needs to move the USB connector there.

ForrestErickson, 2022-07-12:
After review of 20220712, 
Lee removed the Nano foot print.  Fix resulting DRC errors.
Added Test point to UNO foot print for 3v3 pin.  With this change, the GPAD will not supply 3v3 to a shield but we could wire in a 3.3V supply if we really need to.
Changed Test Point names to reflect function two locations.
Minor schematic clean up for readability.

12:22 or so Lee pushed changes to github.

ForrestErickson, 2022-07-12:
While messing with a practice project migrating from KiCad 5 to 6 I added several of the USB parts from the KiCad 6 library to a schematic and a PCB. None of them had 3D models.  Four of them had through hole pins to give them mechanical strength against cable forces.  Two Amphenol, one Molex and One Wuerth

Also found a buzzer with schematic, PCB and 3D model.  TDK : PS1240P02BT and in stock at Digi-key
https://www.digikey.com/en/products/detail/tdk-corporation/PS1240P02BT/935924

[image]

Found 3D model for Worth: 629105150521
In stock at Digi-Key. https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/629105150521/5047751

Here is the 3D view:
[image]

ForrestErickson, 2022-08-02:
We reviewed on 19 and 26 July and no further required changes identified.
The Version 1 PCB design proposed week ending 20220729 has footprints and 3D models for most parts except the JST connector which has it's own issue.

Let's close this today 20220802.

closed 2022-08-02 by ForrestErickson