PubInv/general-purpose-alarm-device #24: Identify Power From Poly Vent to GPAD, 12V or 5V at TBD mA

ForrestErickson, 2022-07-05:
The RJ12 connector in the schematic for the GPAD has both the SPI interface as well as power and ground.
Thus power _could_ be provided to the GPAD as well as the data by one and only one cable.
If 12V is supplied then the power should be labeled as such and the schematic connect to the input of the voltage regulator. Else the power will go (through a fuse and a Schottky diode  at the output of the voltage regulator.

To help choose a fuse, what might be the current that the Poly Vent can supply to the GPAD, 1000mA at 12V?

RobertLRead, 2022-07-05:
Yes. We currently have 24V power supplies that can easily power 500mA (so 12 watts). I think you can assume 1 amp is a baseline at 12V. This is a generally accepted power level, and also can easily be met by an RJ12 connector and cable (if you buy a 23AWG cable.)  I'm only 90% confident of these assertions, but this is a good start.

ForrestErickson, 2022-07-12:
20220712 discussion. Lee identified that at 24V in the linear regulator is likely to be way hot. Better to supply with 12V.

LokiMetaSmith, 2022-07-15:
Per our discussion, 5V is availalbe from the Poly Vent, and since all of the alarm logic signals are 5v, that should work great.

LokiMetaSmith, 2022-07-19:
[image]
move ControllerVcc to the other side of the jumper and add +5v to the other side of the jumper as shown in the image

ForrestErickson, 2022-07-25:
20220725 Lee thinks this is ready to close out.

LokiMetaSmith, 2022-07-27:
This has been accepted and done

closed 2022-07-27 by LokiMetaSmith