PubInv/general-purpose-alarm-device #25: Simulate SPI Controller Peripheral Example Firmware From Repository, nCS on pin D10 and D7

ForrestErickson, 2022-07-05:
Simulate the **SPI_Controller_And_Peripheral** firmware at:
https://github.com/ForrestErickson/SPI_Controller_And_Peripheral/tree/bde15454bd6f0ddee013070e940d8cc6175c0008

Simulate with the nCS (Not chip select) connection on both pin D10 (The example in the code) and on pin D7 as per the proposed schematic of the GPAD as of 20220705. 

Verify correct operation against hardware.

Share simulation with team as a presentation.

Develop and commit simulation to project under ?Firmware? or suggest a location.

LokiMetaSmith, 2022-07-15:
What does the poly-vent side look like? How far along is it? Can we test directly against the  ventilator or do we need to code up a firmware test suit that runs through all possible alarm messages.

RobertLRead, 2022-07-15:
The PolyVent side does not have a well developed fault detection, and if it
did, it would not be easily to mechanically induce all of the faults. We
definitely need a test suite that creates artificial faults. In fact, I
would say we should develop a test suite that has nothing to do with
PolyVent. If needed I can write code for that pretty quickly.

LokiMetaSmith, 2022-08-16:
Add to the digikey order, rj-12 breakout board to support  this needs to have all six pins and six conductors brought out.

ForrestErickson, 2022-10-25:
We found it is not possible to simulate in Wokwi a system with two UNOs (or other controllers) at a time.

With the Version 1 hardware Lee discovered that the SPI Peripheral must have the nCS on pin D10.  We intend to create a version 2 PCB. The Version 1 PCBs are being reworked for nCS on D10.

ForrestErickson, 2025-05-06:
There is a way to implement a custom chip in Wokwi.
This feature may allow programing of an SPI controller as a custom chip to test the GPAD peripheral function.

Watch this video to get an idea and see an example of an SPI peripheral programing:
https://www.youtube.com/watch?v=yzdCS3A4DvU

ForrestErickson, 2025-05-06:
@RobertLRead ,
@LokiMetaSmith ,
@nk25719 ,

See the video in the comment above for very very interesting opportunity to expand Wokwi know how.

Lee