PubInv/general-purpose-alarm-device #27: Simulate I2C LCD interface, A WokWi Example

ForrestErickson, 2022-07-08:
Another starting point for simulation of GPAD functions in WokWi can be found at this project:
https://wokwi.com/projects/333991258811794002

Note: The I2C wiring corresponds to the GPAD wiring.

HOWEVER, the display in the simulation is a 16,2 display so some changes will be required to match the specification for the 20x4 display.

LokiMetaSmith, 2022-07-08:
they have a 20x4 display component. wokwi-lcd2004

LokiMetaSmith, 2022-07-08:
I also started a Simulation directory, that I'm working on

LokiMetaSmith, 2022-07-08:
Alright, I have a working example saved in:
https://github.com/PubInv/general-alarm-device/tree/main/simulation

and can be directly edited here:
https://wokwi.com/projects/309829489359061570

RobertLRead, 2022-07-08:
Good show! I think this is really impressive as a Hello World type
situation. I hope to do more of this soon.

LokiMetaSmith, 2022-08-07:
The electrical interface for the connecting device is described here. https://gitlab.com/polyvent/polyvent_control_module/-/tree/master/

ForrestErickson, 2022-10-25:
The factory test has done this and more.