PubInv/general-purpose-alarm-device #28: Current Rating on RJ Connectors

ForrestErickson, 2022-07-08:
For our review. 
After thinking about fusing of the power supplies on the GPAD I became concerned about current rating on the RJ connectors. 
I found that for POE applications they were limiting to 350mA ?per contact? such as at
https://en.wikipedia.org/wiki/Power_over_Ethernet.

I found no current rating on the RJ12 jack whos part number is: 
https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/5555165-1/769566

However the specification for the plug I found at:
https://www.newark.com/mh-connectors/mhrj126p6cr/rj12-modular-plug-6-position-1/dp/14J4175
allowed up to 1.5 Amps.

Bottom line it looks as if RJ connectors will handle 1A with some margin.

LokiMetaSmith, 2022-07-08:
The buzzer and display should be fine, and the LED's should in general be sufficient. but it's definitely something to keep in mind. If need be, we can double up the number of pins, move to an 8 pin connector (RJ-45) and use two pins for power and ground.

ForrestErickson, 2022-07-08:
Regarding, "If need be, we can double up the number of pins, move to an 8 pin connector (RJ-45) and use two pins for power and ground."  

FYI, I had originally suggested the RJ12 as not likely to be confused with an ethernet connection since they are mechanically not interchangeable.

ForrestErickson, 2022-07-12:
20220712 review, Going with the RJ12 at this time so close issue.

closed 2022-07-12 by ForrestErickson