PubInv/general-purpose-alarm-device #29: Enclosure Design

ForrestErickson, 2022-07-08:
Kincheloe, Lawrence
SPECT

I found this project for designing and 3D printing parametrically defined project boxes. 
https://hackaday.com/2018/03/02/printed-it-custom-enclosure-generator/
https://www.thingiverse.com/thing:1264391/files

It looks comprehensive and perfect for our application.

ForrestErickson, 2022-07-12:
20220712 discussion around enclosure. By moving LEDs from under the LCD to the right of  room can be made for fitting in the USB connector and circuits bringing all  connections to the bottom edge.

LokiMetaSmith, 2022-07-18:
I pushed a folder in the Hardware directory for the Enclosure, that contains the latest STEP 3D export of the PCB and an initial mock-up of the enclosure, as well as a thrown together FreeCAD file that imports both in a single view.

LokiMetaSmith, 2022-07-18:
Hum,. I somehow obliterated everything I was working on over the weekend and today. Not sure how that happened. I'll have to re-build what I was working on for the enclosure.

LokiMetaSmith, 2022-07-19:
https://medium.com/@livnoorbrar/lost-code-by-accidentally-creating-a-detached-head-dont-worry-58919f32b08d

This saved my butt. It should be safely committed now.

LokiMetaSmith, 2022-07-19:
Add mounting tabs to the enclosure, per discussion with Lee

ForrestErickson, 2022-07-26:
I have created a mechanical placement drawing for (I hope) all of the components that penetrate the enclosure. 
For the LCD I have dimensioned to the center. It is not very obvious.
For the UNO footprint I have dimensioned to the pin 1 and none of the other features.

The file is at: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB/MechanicalPartsPlacement.pdf

ForrestErickson, 2022-07-26:
Added drawings to parts folder for potentiometer and for JST-MFG connector.

LokiMetaSmith, 2022-08-09:
This is taking longer than expect, work is going well. Translating the origin coordinates was non-obvious at first.

LokiMetaSmith, 2022-08-22:
[image]

coming together

ForrestErickson, 2022-10-25:
As of 20221025 Robert has received a prototype enclosure from TreatStock and will report on fit next week. Including how well the screws drive.
Photos when possible.

RobertLRead, 2022-11-01:
This enclosure is accepted as a workable design at present, so we are closing this issue.

closed 2022-11-01 by RobertLRead