PubInv/general-purpose-alarm-device #30: "HandSoldered" SMT Footprints versus Standard SMT Footprings

ForrestErickson, 2022-07-12:
I left the following question at the KiCad Facebook group

_While learning my way into KiCad I found there are footprints for "HandSoldered" SMT parts like this 0603 resistor. 
I assume the down side to using the "HandSoldered" footprint in an SMT production is greater chance of tombstoning.
I have often hand soldered first samples of an assembly and it is tough work and I seldom get the parts well centered.
So any experience out there?_
[image]

ForrestErickson, 2022-07-12:
Got some responses like this

- Hand solder footprints are great for production. They’re ideal and allow for substituting a 0805 part as needed.
- I used these for my first hand built SMD board. But on my second there was no room, so I used the standard ones. However, I had no issues hand soldering on any of them, so I don't use the hand solder ones any more.
- Same, I normally hand solder the first prototype. Using a standard footprint, it doesn't make it too difficult to solder as I use 0603 as my standard.
- Recommended use a standard footprint for space efficiently and maybe order the Stencil to make your hand solder easier.
- I usually use handsoldered size pads. They work great with hot air gun, oven, iron, whatever. The surface tension of the solder pulls the part to the center of the footprint with hot air and it doesn’t tombstone if you heat it correctly.

Finally this tip seams worth remembering:
Tombstoning typically occurs when one side is heated up quicker (or only) than the other, and if the component isn't secured down, so it just depends on your SMT soldering method.

Lee thinks he may use the "HandSoldered" SMT Footprints.

LokiMetaSmith, 2022-07-15:
This board isn't a tight design, and it's not a high volume production design, so hand placed resistors are fine.

ForrestErickson, 2022-07-29:
I have edited the foot prints for the resistors and capacitors for hand surface mount.
[image]

closed 2022-07-29 by ForrestErickson