PubInv/general-purpose-alarm-device #31: Tact Switch Cap, Choice / Design

ForrestErickson, 2022-07-13:
For the GPAD there is a user button next to the buzzer.
Manufacturers with stock at Digi-Key  include TE Connectivity and Omeron
These switches are designed for a Cap or Button which snaps onto the moving part of the switch.
We must find a cap with a compatible height or design one.

For Omeron
For button see:  https://omronfs.omron.com/en_US/ecb/products/pdf/en-b32.pdf
Buttons up to 12x12mm. are available.
https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/B32-2010/1811718
Manufacturer Product Number: B32-2010
For switch see: https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/B3F-1152/368364
Manufacturer Product Number: B3F-1152

For TE Connectivity
Example button will be a 6x6mm tactile switch (tact switch) such as  450-1655-ND
The drawing for which is: https://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=1825967&DocType=Customer+Drawing&DocLang=English
I can not find a drawing for the key caps at this time.  I have sent an email to Digi-Key asking for help finding cap drawing and PN.

As of 20220713 the part in the Schematic does not have a cap retention geometry and has only short button:
- Distributor 1	JLCPCB	0	Center	Center	0	0	1.27 mm	
- Distributor 1 PN	C13828	0	Center	Center	0	0	1.27 mm	
- Distributor 2	Digikey	0	Center	Center	0	0	1.27 mm	
- Distributor 2 PN	450-1804-ND	0	Left	Center	0	0	1.524 mm	
- Manufacturer	TE Connectivity ALCOSWITCH Switches	0	Left	Center	0	0	1.524 mm	
- MPN	TE Connectivity 1825910-7,   Dongguan Guangzhu Industrial C13828	0	Left	Center	0	0	1.524 mm	

The length of the switch and the length and size of the cap will be required for design of the enclosure. Must be compatible with the height of the LCD module too.

LokiMetaSmith, 2022-07-19:
Migrate to 12x12mm for the mute switch

ForrestErickson, 2022-07-25:
The Enclosure Design issue: 
https://github.com/PubInv/general-alarm-device/issues/29 [Hardware](https://github.com/PubInv/general-alarm-device/issues?q=is%3Aissue+is%3Aopen+label%3AHardware)
#29 should be made dependent on this.

ForrestErickson, 2022-07-25:
The switch in the schematic / PCB for the alarm mute is Omron Electronics B3F-4055.
https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/B3F-4055/31799

This switch takes key caps such as:   B32-1310
https://omronfs.omron.com/en_US/ecb/products/pdf/en-b32.pdf
See drawing for full mechanical.
Example: 
[image]

This detail shows a 10mm dimension from the PCB surface to the top of the button. and suggestes the enclosure be designed for 1-2 mm of key rise above the enclosure surface.  So that leaves 8mm for the GPAD pcb to outside surface of the enclosure. 

The header which will connect the LCD to the GPAD PCB has a thickness of 0.1" so we will want a 0.125 ?nylon? spacer at the four mounting holes of the LCD to PCB. 

Excluding the opening for the LCD the board that mounts the LCD will have a top surface high point of the header pins which measures 0.128 + the 0.125 spacer of about 0.25" or about 6.35mm. 
This would allow for something like 8-6.35 = 1.65 mm (or about 0.065") enclosure thickness. 

Added https://grabcad.com/library/tactile-switch-thru-12mmx12mm-with-cap-1  to Schematic symbol and PCB footprint .
Pushed to repository at about 13:00 EDT.

ForrestErickson, 2022-07-25:
I have inserted the switch with button model into the PCB.
Top view.
[image]

Bottom view.
[image]

**CAUTION the LCD needs yet to be spaced away from the PCB by 0.125.**

LokiMetaSmith, 2022-07-27:
This feature is acceptable, We may create an issue for the enclosure that references this issue for a custom key cap

closed 2022-07-27 by LokiMetaSmith