PubInv/general-purpose-alarm-device #32: Layout Change List

LokiMetaSmith, 2022-07-19:
We'll use this issue to list all the layout changes needed.

LokiMetaSmith, 2022-07-19:
Move the uno by rotating it 90 degrees such that the connectors come from the bottom.

Move the interfering LEDs vertically such that they are to the side of the LCD display.

Move the reset button and LCD contrast potentiometer to the side of the board with the LCD

ForrestErickson, 2022-07-24:
20220724 0724 push changes to main.
Note I am not sure of the centering on adjustment screw on the silk screen of the contrast control pot.  
Should the POWER LED move up and perhaps the contrast control pot move down?
PCB screen shot:
[image]

3D render Front aka component side:
[image]

3D render Back aka Display side:
[image]

ForrestErickson, 2022-07-24:
Oops, J105 is not on grid. Will need to fix once the schematic loads again.
[image]

ForrestErickson, 2022-07-24:
Fixed the library and schematic error and tweeked the location of J105.

At 16:09 EDT I have pushed the change to github.
Ready for review.

ForrestErickson, 2022-07-24:
The centering on the contrast POT is hard to understand.
The relation between axis where the tool turns the pot and pins and the component outline is more than I can understand this late in my day.

Here is a drawing for the 3/8" part: https://www.bourns.com/docs/Product-Datasheets/3386.pdf

ForrestErickson, 2022-07-25:
20220725 Morning of, I found a 3D model for Arduino UNO on grabcad and made a footprint for it and pulled it in. Good thing too because it made me realized **I have to flip the UNO footprint** from PCB top to bottom to be correct.

Will need to allow for about 0.350 of female connector height and 0.100 of pin header height.
These settings get the PCB flush.
[image]

For an offset of -.575 which looks like this.
[image]

Here it is after putting the footprint on back.
[image]
and
[image]

I rotated the uController and near buy circuits too.
[image]

ForrestErickson, 2022-07-25:
New push as of 20220725 10:11 EDT.

ForrestErickson, 2022-07-25:
Possible Spacer for LCD from PCB
https://www.mcmaster.com/94639A702/
Shape	Round
Material	Nylon 6/6 Plastic
OD	3/16"
OD Tolerance	-0.015" to 0.015"
Length	1/8"
Length Tolerance	-0.015" to 0.015"
For Screw Size	No. 4

We will need a screw long enough to go through 2x 0.0625 + 0.125 + thread engagement of 0.125 = 0.375
This part looks good:
https://www.mcmaster.com/90272A108/
Zinc-Plated Steel Pan Head Phillips Screw
4-40 Thread, 3/8" Long
Threading	Fully Threaded
Head Diameter	0.219"
Head Height	0.080"
Drive Style	Phillips

We will need nuts.
Steel Locknuts with External-Tooth Lock Washer
https://www.mcmaster.com/90675A005/
Material	Zinc-Plated Steel
Thread Size
4-40
Thread Type	UNC
Thread Spacing 
Thread Fit	Class 2B
Thread Direction	Right Hand
Width	1/4"
Height	7/64"

Shopping cart totals less than $18 exclusive of shipping.
[image]

ForrestErickson, 2022-07-25:
All parts affecting enclosure on PCB locked with commit pushed at 20220725 about 16:55 EDT.

LokiMetaSmith, 2022-10-04:
Closing this as the V1 boards have been built up.

closed 2022-10-04 by LokiMetaSmith