PubInv/general-purpose-alarm-device #33: Understanding Tagging on Git and Github

ForrestErickson, 2022-07-23:
With out fear of contraction Lee can say he does not understand tagging work flow on git CLI and Github web interface.

[image]

I wanted a tag before I began a tear up of PCB traces and made the above local tag and pushed it.

I see no new tag on the web interface as I expected I would.

[image]

LokiMetaSmith, 2022-07-24:
Sharing Tags
By default, the git push command doesn’t transfer tags to remote servers. You will have to explicitly push tags to a shared server after you have created them. This process is just like sharing remote branches — you can run git push origin <tagname>.

So following the above, you must simply type:
git push origin VersionZeroPCB 

From: https://git-scm.com/book/en/v2/Git-Basics-Tagging#:~:text=Sharing%20Tags&text=git%20push%20%2D%2Dtags,be%20pushed%20to%20the%20remote.

ForrestErickson, 2022-07-24:
Thanks for the reminder. 

It worked:
[image]

ForrestErickson, 2022-08-02:
Lee thinks he can find these notes on Tagging the next time needed. That time will be when we place the order for Version 1 PCBs as we will want to be able to refer to this hardware build event.

Let's close this today 20220802.

closed 2022-08-02 by ForrestErickson