PubInv/general-purpose-alarm-device #35: Identify and Order Long Lead Parts for Prototype #1 Build, Build Quantity = 10

ForrestErickson, 2022-07-25:
Soon we should identify and order parts for a Prototype#1 GPAD build.
I suggest we order for a quantity of 10 with intent to build 5 and have both spare parts and if all goes well we build up the additional units.  Some exceptions. Order so that we get 5 raw PCBs for mechanical or test and troubleshooting aids.

The orders should be driven by the KiCad BOM and Parts list and the same may need some "maturing" especially with respect to what might be assembled by JLCPCB versus what cannot. 

We will need a Digi-key order for some parts we can not get frmo JLCPCB and the LCD modules from ?Amazon? or ???

Some Digi-key parts in a shopping cart.
https://www.digikey.com/short/wdb03rb9

RobertLRead, 2022-07-26:
Okay, this sounds good. I'm happy to do this with you "pull the trigger". I
leave for the Linux Expo on Wednesday, which may make me a little slower to
communicate.

RobertLRead, 2022-07-26:
The deliverable here is 15, and we will order some spares.

ForrestErickson, 2022-08-08:
Robert, 
Lawrence,

In addition to the parts for the assemblies the RJ12 break out modules need to be ordered in sufficient quantities for all users. So all users need to be identified too.

RJ12 6 Ft (2M) Data cables:  https://www.amazon.com/NECABLES-Short-Cable-Straight-Wiring/dp/B08BHVF6XS?th=1
RJ12 break out module.   See: https://www.amazon.com/SchmalzTech-Screw-Terminal-Breakout-ST-RJ11-R-MIN/dp/B09KQZG58C
12V @ 1000mA power supply with 2.1mm barrel jack.  See: https://www.amazon.com/ZOSI-1000mA-Adapter-Security-Surveillance/dp/B072PZBT8G/ref=asc_df_B072PZBT8G

RobertLRead, 2022-08-08:
> I updated the BOM spread sheet generated from the schematic. The sheet is
> at: GeneralPurposeAlarmDevicePCB.xls
>
> I updated the current cost from the distributors.
> Note I have parts into the schematic for the enclosure and the PCB(and SMT
> assembly cost) for which I have no cost.
> I have added no cost for an Aruino either.
> So there are three parts with no price which I have entered as $1,000,000
> just to call attention to them.
>
> Exclusive of these three parts the cost is $21.28.
>
> Robert,
> Please look at the sheet. Can you order long lead parts from this or do
> you want help generating other formats or catagorizing parts?

I can order from this, though I think on Tuesday (I know you will miss that
meeting Lee), we are going to do this.

In the future it might be useful to put URLs in a spreadsheet for ordering.

I'm a little unsure how to tell a "long lead" part from any other.

Do you mean that these parts are ready to be ordered right now?

ForrestErickson, 2022-08-08:
RE: a long lead is often synonymous with principle components and that is often found by sorting by price.At this point all parts should be ordered and try to capture expected delivery dates becomes the useful way to project when assembly steps can happen.The wrinkle you have is that most parts, but not all, JLCPCB can assemble and that must be tracked so you do not double order or fail to order parts.There is some  stress in managing it.Forrest Lee EricksonCell: 865-518-1040Sent from cell.

  for Prototype #1 Build, Build Quantity = 15 (Issue #35)

LokiMetaSmith, 2022-08-16:
Moving this to the orderable milestone

ForrestErickson, 2022-09-06:
In hind sight Lee sees that "long lead" was not specific enough for others to work with. Perhaps a field in the parts master (parts list, BOM) would help manage this in the future.

closed 2022-09-06 by ForrestErickson