PubInv/general-purpose-alarm-device #36: Add to PCB drawing in User.Drawing the Key Component Locations.

ForrestErickson, 2022-07-26:


ForrestErickson, 2022-07-28:
Lee added on user drawing layer. 
[image]

On 20220727 used in review with Lawrence for enclosure design. Sufficient for enclosure design.

ForrestErickson, 2022-07-28:
Lawrence,
**Note:** Very slight change to U105 centering.
[image]

J105 is now at 3200.0 exactly. It was at something 3200.0070
[image]

ForrestErickson, 2022-07-28:
Lawrence,
Please note very slight change on X for J105.

ForrestErickson, 2022-07-30:
Is there a way to get a DXF or similar file for these locations?
Something that could be imported to SCAD or FreeCAD rather than re entering manually the data?
The diameter of the LEDs would need definition and similar for all other enclosure apertures.

LokiMetaSmith, 2022-08-04:
This appears done, might need to readdress when case layout is finalized

ForrestErickson, 2022-08-04:
Let's close out and make a new issue if the enclosure design requires a change or clarification.