PubInv/general-purpose-alarm-device #39: Design for Manufacturing Test / Troubleshooting

ForrestErickson, 2022-07-26:
**DFM DFT,** 
Review the GPAD.
Design for Manufacture, Design for Test

- Add test point to all power supplies with labels.
- Decouple all loads so current can be measured.
- Check for access and clearance around all mounting holes.
- Check for instructions for manufacturing and user text on PCB.

**Items when we Write Manufacturing Test / Troubleshooting Procedure**
Place and attach printed wiring assembly into enclosure display side leaving cover off.

For factory test and for trouble shooting.
Record DUT Serial number (assign serial number if first test.)
Begin with no firmware. Measure and record the following.  Compare against limits and troubleshoot as required.
1. Begin with no power supplied.
2. Apply power at USB connector J501 . Measure Supply at all power Test Points at normal loads. Measure current through each decoupling resistor. Disconnect at J501.
3. Apply power at J101. Measure Supply at all power Test Points at normal loads. Measure current through each decoupling resistor. Disconnect at J101.
4. Apply power at J101. Measure Supply at all power Test Points at normal loads. Measure current through each decoupling resistor. Disconnect at J101.
5. Load firmware into DUT and remeasure voltage at all TP and current through all decoupling and capture.
6. Save measurements.  To where?
7. Verify all features fully functional.   How?
8. 

Finish assembly.
Apply serial number label.
Final light and play after last screw driven aka, test for IWT (Intermittent with tap) and RWT (Rattle with tap).

LokiMetaSmith, 2022-08-04:
Apply power to J101 first with a current limited power supply. Set the current limit to measured standard  no load current draw up to max current draw. This allows the board to power on safely but catches an over current event without causing long lasting damage to the board in most cases.

LokiMetaSmith, 2022-08-16:
This issue has been agreed to be closed.

closed 2022-08-16 by LokiMetaSmith