PubInv/general-purpose-alarm-device #40: As a user, I need the GPAD to have a sturdy enclosure that allows me to attach it to a medical device, such as a with zip ties.

LokiMetaSmith, 2022-07-28:


LokiMetaSmith, 2022-08-04:
Need to add zip tie mouting hole options to the enclosure

ForrestErickson, 2022-10-11:
Further versions will add mounting details.

ForrestErickson, 2022-10-25:
The version 1 enclosure does not have such mounting provisions excepting perhaps through the vent slots.

RobertLRead, 2022-11-01:
We still need this feature for the next enclosure design Enclosure 2.

LokiMetaSmith, 2022-12-15:
We could use some solicited feedback on how people want to mount it. One way to quickly prototype would be to screw a large plate to the back of a case, and make holes where we need them to get an idea how best to add mounting holes.

RobertLRead, 2023-01-03:
@LokiMetaSmith I don't think we should get feedback here; we should just add some holes in the enclosure for zip ties. That gives all the flexibility we need. If you disagree, let's talk about it next meeting; if you agree, let's just modify the enclosure.

ForrestErickson, 2023-05-11:
@RobertLRead ,
@LokiMetaSmith 

Regarding the requirement for "sturdy enclosure".  The V105 enclosures are not sturdy. This. "sturdy enclosure" remains a very interesting development to under take.

ForrestErickson, 2023-06-20:
Version 2 has been attached to a bed railing. DONE.

closed 2023-06-20 by RobertLRead