PubInv/general-purpose-alarm-device #41: As a user in a Low and Middle Income Country (LMIC), I need GPAD to cost less than $100USD.

LokiMetaSmith, 2022-07-28:


LokiMetaSmith, 2022-07-28:
This requires an estimated BOM to be generated

ForrestErickson, 2022-08-02:
Lee has generated a costed BOM. See file: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB.xls

The KiCad BOM CSV file is opened and pasted into a tab in this spread sheet which has another tab to process for cost estimation. There is an instruction tab too.
 
There are three components for which no price is known. The PCB and assembly cost, the enclosure, and a TBD Arduino shield for applications TBD.  Lee entered the cost for these three items as $1,000,000 to call attention to them.
Also not captured or flagged, is cost to test nor any packing material.

The resulting cost is $3,000,0027.28.
Screen shot:
[image]

ForrestErickson, 2022-08-02:
We need estimates for PCB and assembly. Assuming this will come as we generate a purchase from JLCPCB.
We need an estimate for the enclosure.
Lee thinks we can set the hypothetical Arduino UNO shild cost to zero for now.
Do we want to add in estimates for test (including loading firmware)?
Do we want to add in estimates for packaging?

All of this is reoccurring costs. No estimates for one time costs. No one time costs (such as test for FCC Part 15 EMC) are identified as required.

LokiMetaSmith, 2022-08-02:
Assembly estimated to be $15/hr and maybe 15-30 minutes to test, assemble and verify.
the enclosure can be estimated (when it is complete) with Shapeways.com, or pcbyway's 3D printing service. It is estimated that the cost of  the enclosure will be below $15

Also, the arduino is for development purposes and won't be included in the final BOM

closed 2022-08-02 by LokiMetaSmith