PubInv/general-purpose-alarm-device #44: Make 3D Body for JST Connector at J105

ForrestErickson, 2022-07-28:
Lee has made a PCB footprint but has not made a 3D foot print. 

Lawrence, Suggestions?  

PCB screen shot.
[image]
Connector screen shot.
[image]

Connector data sheet:
https://datasheet.lcsc.com/lcsc/1811011927_JST-Sales-America-S04B-XASK-1-LF-SN_C265097.pdf

LokiMetaSmith, 2022-08-04:
No CAD looks available, recommend square 3D block to represent the connector

ForrestErickson, 2022-11-13:
Updates PCB and Pretty files committed with simple rectangle made in FreeCAD for 3D model.

@LokiMetaSmith Could you update your PCB and try the fit into the enclosure to check for the JST connector. Note the latch on the jack to which the plug engages is not modeled.

ForrestErickson, 2022-11-13:
Top view
[image]

[image]

closed 2022-11-22 by LokiMetaSmith