PubInv/general-purpose-alarm-device #45: Adding Hardware to PCB Assembly such as Header, Screws, Spacer and Nut for LCD

ForrestErickson, 2022-07-28:
Lee asked for the following advise 

> Advise please on how to deal with mechanical parts. 
Specifically I have an LCD module that will require a 16 pin header to connect to my PCB. Also four nylon spacers and four 4-40 screws and four #4 lock washers.
I think I want to add parts on the schematic with no electrical connection. 
What might be the pit falls?
I went looking into the installed schematic symbols for "hardware" and "Washers" with no success.
Here are the parts I intend to use:
I have a sub sheet with the LCD where I intend to put the symbols.

and received this very interesting work flow description:

> [Emery Premeaux](https://www.facebook.com/groups/861083273995933/user/696647783/?__cft__[0]=AZWU8t4S2EZeaHy3F1-GlHkT-60mLphSNR-a88XAIXgcAxP9qGlqT5VpsovTnAqrbhjIFw2wyVDVKSoi6JbCx0ttuA_51phW9Ye6ZOes28erkNS71hM2epkgRBat-mth2eFICP0yVuk04JGuq3_WdpR8aEv4-1cy1l-NTjjdK1bd3c4oVI9aOCAmYmQ4Wrr1r08&__tn__=R]-R)
> I use two methods depending on the part:
> 1: make symbols for the mechanical part and put on sheet. Like on OSH logo on sheet represents a silkscreen "footprint" logo on the pcb. For example, i have a drawing of a pcie card which is not an electrical component, but loads the cad file of that card to be placed on my carier board. This method is great for all of these little modules from sparkfun, adafruit, etc.
> You could draw a screw symbol.
> As a footprint this could be a NPTH hole, or just a footprint with only fab layer data drawn. So draw the footprint in the FOOTPRINT EDITOR like youd make any other component for your global libraries.
> Finally on the footprint edit the settings, on the 3d tab load your step file and position it. Save. Load footprint to your board. And position.
> 2: this method just adds a 3d step file to a footprint/3d part. Typically when I dont ALWAYS want that mechanical part attached to a base item (such as a screws for an lcd screen. On THIS project you will use screws, but not every project with that LCD needs screws).
> Do your normal proceedure of schematic and footprint association for the LCD.
> In the PCB LAYOUT edit the settings of your primary part (LCD screen for example). Go i to the settings/3d tab, and just like before, add the step files for screws and posts, etc and position as required. Notice that you can add MANY step files and show or hide them with the check box. So you can have the lcd step, several instances of screws, posts, heat sink, fan etc.
> When ready, save to the BOARD. This only modifies the project copy of the base footprint. Not your global library instance.

If I understand correctly either of These would give us 3D parts in the board to view, render, the 3D view more complete. 

While not essential to what we are doing it would be aesthetically gratifying.

ForrestErickson, 2022-07-29:
20220729 11:44 or so, Lee committed changes adding a PCB and enclosure to the schematic and has updated the spread sheet GeneralPurposeAlarmDevicePCB.xls to produce a BOM for purchase of parts and cost estimate. Note there are three parts with no price entered as $1,000,000 just to call attention. 

The BOM file is: [GeneralPurposeAlarmDevicePCB.xls](https://github.com/PubInv/general-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB/GeneralPurposeAlarmDevicePCB.xls)

Lee has not sorted out by distributor.

LokiMetaSmith, 2022-08-04:
This looks good, There's some impressive spreadsheet magic going on in the BOM file. 

More than sufficient to hand off to a contract manufacturer for assembly

LokiMetaSmith, 2022-08-30:
This should include the UNO arduino headers.

LokiMetaSmith, 2022-10-04:
THis has been done for Rev1

closed 2022-10-04 by LokiMetaSmith