PubInv/general-purpose-alarm-device #46: 20x4 LCD Footprint Error (Warning)

ForrestErickson, 2022-07-29:
While it should be a warning, KiCad is set up to return an error it has found in the LCD footprint .

[image]

This is a detail of the court yard was was inherited from the PCB footprint (which I got I know not where any more).
Perhaps I made it from a 20x2 LCD. I did not draw this feature.

While I can and will short term, set to ignore this error I will try to find time to understand the root cause and modify the foot print to clear it in the future.

LokiMetaSmith, 2022-08-04:
This issue isn't showing up when I run DRC.

ForrestErickson, 2022-08-04:
Regarding, "This issue isn't showing up when I run DRC."

I have probably added it to the exclusion list. 

The delete all markers check box may bring up a dialog for finer grain clearing of DRC exclusions.  I do remember once being confused about it all and unable to get back to errors I know I had excluded. I attributed it to my old brain not remembering how to find the dialog but perhaps there is a bug.

ForrestErickson, 2022-11-13:
Notes on clearing this error.
I made a new simple project. I selected an UNO shield template and called the project KiCadLibrary. 
[image]
Using the Symbol and Footprint library managers, I then connected the new project to the libraries in the GPAD folder.

[image]
[image]

I added the LCD to the schematic and annotated it.
I imported new parts into the PCB.
I ran the DRC and reproduced the errors related to the LCD. Courtyard not closed and self intersecting.
[image]

I right clicked to edit the footprint and turned off all but the front courtyard layer.
[image]

I selected the suspect lines and moved to the front silk screen layer.
[image]
Saved the change and reran DRC.

This cleared the errors.
[image]

ForrestErickson, 2022-11-13:
The work above was only on the local PCB copy of the foot print. So reopened the footprint editor and used Saved As to save into the GPAD library.

[image]

And selected to overwrite.
[image]

I added a second LCD to the schematic and updated the PCB to bring it it. I reran DRC and there were no errors for the second LCD.

ForrestErickson, 2022-11-13:
I think this clears this issue with the LCD module. 
The interfering mounting holes remain to fix.

ForrestErickson, 2022-11-13:
Note on LCD mounting holes. The LCD measures 0.133 inch diameter or 3.37 mm.
The pad property on the Version 1 PCB footprint is only 2.5mm
[image]

Change to:
[image]

Make holes through hole plated and ground. Add to PCB.
[image]

closed 2022-11-22 by LokiMetaSmith