PubInv/general-purpose-alarm-device #47: Route GPAD PCB Prototype Version 1

ForrestErickson, 2022-07-30:
Amusing that no issue was made for this. 
Lee has committed a routed PCB as of about 20220729 20:30

While Lee can see improvements, it is ready for review as is.
(If we re ordered the pins driving the LEDs we could reduce the signals crossing over each other.)

Would like other eyes on it.

ForrestErickson, 2022-07-30:
On 20220730 Lee is reviewing the silk screen placement.

ForrestErickson, 2022-07-30:
20220730 9:00 Lee pushed silk screen changes. Done with silk screen review.

LokiMetaSmith, 2022-08-04:
Reviewed PCB layout, 

- Footprint, orientation, and arrangment of J105 looks ok.
- Q602 is missing 3D model
- XTAL101 tracing isn't a tuned differential pair and trace length differs
- only one via size used, even with different trace path widths
- no via stitching and ground islands, while all connected, in many cases only have one via
- Several areas where ground pours could be eliminated
- All design rules passed
- F101 is sufficiently sized for it's use case, 1A nominal/8A max below 0.5 ohm in line resistance
- Trace width changes drastically in some areas, some of it looks intentional, and it's assumed it's to improve reliability in areas.
- Part placement and alignment is sufficient for hand assembly, silkscreen designators are easy to read. 

Recommend closing this issue, as board seems sufficient for stated goals of v1

ForrestErickson, 2022-08-04:
This is just FYI:   Regarding, "XTAL101 tracing isn't a tuned differential pair and trace length differs"  

Traces to pins on a crystal create parallel capacitance. For a crystal in a Voltage Controlled Oscillator (VCOs) such as in a PLL application such parallel capacitance reduces the pull range of a crystal and may need to be taken into account when setting the capacitors which are part of the tuned circuit. For VCOs it is common to remove ground under the crystal and under the traces.  Traces are made thin to reduce capacitance and placed "far" apart from each other.

For our fixed frequency application none of these design constraints apply. 

Times have changed and even with googeling for "vcxo design notes" I found nothing helpful to reference.

ForrestErickson, 2022-08-04:
Regarding, "only one via size used, even with different trace path widths"

Dang. It had been my intent to use large vias on the power traces (set to 0.05') and I think I did on the version zero PCB. But forgot about it on this layout. I think there are net class settings for vias in to which I should look and see if I can change my settings so that future PCBs will follow this intent automatically. 

For the GPAD there are no high currents where the small vias are going to cause an important voltage drop and so we can use as is. 

Thanks to Lawrence for pointing this out.  Lee will make an issue to try to set up KiCad for enlarged Vias on power nets.

ForrestErickson, 2022-08-04:
Regarding, "Several areas where ground pours could be eliminated"

FYI, I have always practiced adding ground pours where they can be with out causing problems because I was told it causes less copper sulfate handling (and waste opportunity) at the PCB fabrication process.

ForrestErickson, 2022-08-04:
Lee recommends closing this V1 issue too.

ForrestErickson, 2022-08-04:
Regarding, "Times have changed and even with googeling for "vcxo design notes" I found nothing helpful to reference."

While not as well explained, this is close to the application notes that helped me years ago better understand VCXO design and shows specifically how the Shunt capacitance (what I was calling parallel capacitance) harms pull range.  It is a very useful thing to know should you ever be in the position to be designing a VCXO.  It is also _very interesting_.

https://www.crystek.com/documents/appnotes/VCXOarticle.pdf

ForrestErickson, 2022-08-04:
Gerbers and drill files generated and zipped. File name: GerbersV120220804.zip

Checked at both JLCPCB and PCBway Gerber viewers and look OK.

[image]

Check at PCBWAY for holes on UNO footpring.

[image]

LokiMetaSmith, 2022-08-16:
This has been completed to all parties satisfaction.

closed 2022-08-16 by LokiMetaSmith