PubInv/general-purpose-alarm-device #48: Update to Main Readme.md To Reflect Project Progress

ForrestErickson, 2022-07-30:
The main Readme.md at https://github.com/PubInv/general-alarm-device#readme is getting dated.

Our status (progress) should be more on display.

Summary of progress.
Firmware proof of concept for SPI Peripheral (and Controller)
Proposed Prototype Schematic
Proposed Prototype PCB

The issues defining features should be collected some how to form a product definition

ForrestErickson, 2022-08-08:
Robert,
Joe,
Lawrence,

I have made a branch and pushed to the repository at: https://github.com/PubInv/general-alarm-device/tree/UpdateMainReadme%2348

It very likely has my "novel spelling" all over it. 
I was concentrating on changing content to reflect what I think is our status.

There is a section where I was indicating our roles and I was with out an idea to describe how Joe and Lawrence view their role nor did I put in a role for Speccollective.

I hope you can review it and accept or offer changes before I leave for WI and MN on the morning of 10th.
I might have a window of opportunity the evening of the 9th.

Robert,
I have summarized near the end the license but worry it should be more elaborate.

The way forward might be to accept what I have proposed today 8 August 2022. Let me merge to main and push and then one or more of you could fix it up further.

LokiMetaSmith, 2022-09-13:
We've created the initial readme and reflects the status of the project at that time.

closed 2022-09-13 by LokiMetaSmith