PubInv/general-purpose-alarm-device #50: Try to Set Up KiCad for Enlarged Vias on Power Nets.  (Ground too)

ForrestErickson, 2022-08-04:
try to set up KiCad for enlarged Vias on power nets.
See notes on Route GPAD PCB Prototype Version 1 #47

LokiMetaSmith, 2022-08-16:
This issue should be left for version 2.

ForrestErickson, 2023-01-03:
Still a to do as of 20230103.

ForrestErickson, 2023-02-21:
Notes.
Under board setup the net classes can be set for things like trace width and via size.

Here are the current settings as of 20230220.
[image]

Some notes on DC resistance  calculations.
Using the idea that 1 oz copper has 0.5 mOhms per square the via with a diameter of 15.7 mills has a circumference of about 49 mills and a length of 62 mills and so is just short of square and will have a bit less than 0.5 mOhms.

ForrestErickson, 2023-02-21:
Under Edit there is a Set Track and Via Properties dialog. There is a Set to a specified vault dialog options.

[image]

ForrestErickson, 2023-02-21:
Lets not risk change at this date to the Rev 2 PCB. 
Will close as Will Not Fix.

RobertLRead, 2023-06-20:
Closed as wont fix.