PubInv/general-purpose-alarm-device #53: Integrate the two simulation files that we have to create an initial version of firmware.

RobertLRead, 2022-08-13:
One of the files is: [https://github.com/PubInv/general-alarm-device/tree/main/simulation/robSimulatinSat13](https://github.com/PubInv/general-alarm-device/tree/main/simulation/robSimulatinSat13)

LokiMetaSmith, 2022-08-16:
#26 needs to incorporate this into the integration as well.

ForrestErickson, 2022-08-16:
Lee has experimented some more with Wokwi.
I have simulated making tones and learned that I was mistaken about simultaneous tones. The Arduino tone() command can only work on one pin at a time. (I am assuming I once saw a DTMF project that made tones another way which lead me to my mistaken conclusion.)

However, with the hardware as implemented in Version #1 PCB we can set a **Tone** pin and we can assert an **Attenuate** pin low impedance OUTPUT to reduce the volume of the **Tone**.   Please note another constraint, that when setting the Tone pin that **Light0** pin cannot be a PWM (Cannot use Analogwrite on Light0 pin).  See https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/
_Use of the tone() function will interfere with PWM output on pins 3 and 11 (on boards other than the Mega)._

Here is a Wokwi simulation file Lee made to exercise the tone and attenuation. 
https://wokwi.com/projects/338616554801332819
[image]

ForrestErickson, 2022-10-25:
In issue #90 we have found we can create Wokwi specific code.
This is closed.

closed 2022-10-25 by ForrestErickson