PubInv/general-purpose-alarm-device #54: Make PDF Files of Schematics and PCB Layers, Verion #1

ForrestErickson, 2022-08-14:
Make PDF files so that persons with out KiCad installed can see schematics and PCBs in their browser PDF viewer.

20220813 Lee is traveling. Uploading KiCad on ancient laptop to try to make the PDFs.

LokiMetaSmith, 2022-08-14:
This has been done, but in retrospect, the PDF needs to be versioned with the major.minor board revision it's related to, and as @ForrestErickson suggested, move it into the PCB folder. 

If we had a block diagram with all of the broken out pins, their names, and their Arduino references, that would go in the Documentation directory, where as the schematic is more appropriate for the PCB design files, as the purpose of the PDF export is to remove the need to open up whatever editor the schematic was first generated in. As an aside, the exporting of a PDF is a commonly overlooked practice in most Open Hardware projects, but is very necessary!

ForrestErickson, 2022-08-14:
> Regarding Roberts question of which pins for which LEDs which started the
> quest for a schematic see the screen shot below.  IIRC LIGHT0 is on the
> bottom and so on.

This is pretty awesome --- can you stick it in GitHub as Lawrence suggests?
(beyond just this issue, I mean.)

ForrestErickson, 2022-08-14:
It is in the git hub repository
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/PDF/Schematic-GeneralPurposeAlarmDevicePCB-V120220810.pdf

LokiMetaSmith, 2022-08-16:
All parties agree that this solution is acceptable

closed 2022-08-16 by LokiMetaSmith

ForrestErickson, 2022-10-11:
Regarding Roberts question of which pins for which LEDs which started the quest for a schematic see the screen shot below.  IIRC LIGHT0 is on the bottom and so on.