PubInv/general-purpose-alarm-device #55: BOM error U302, bad merchant links

LokiMetaSmith, 2022-08-15:
The AliExpress and Amazon link reference the version of the display that has the i2c module presoldered to the display. 

53 | 1 | U302 | LCD_20x4_Character-GPAD_SCH_LIB | GPAD_SCH_LIB:LCD_20x4_Character-GPAD_SCH_LIB | GeneralPurposeAlarmDevicePCB:LCD_2004A |   |   | HAND |   | 4.99 |   |   |   |   | Aliexpress | https://www.aliexpress.com/item/3256803213374992.html | Amazon | https://www.amazon.com/GeeekPi-Interface-Adapter-Backlight-Raspberry/dp/B07QLRD3TM/ref=sr_1_2
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

LokiMetaSmith, 2022-08-15:
This looks well priced. 

https://www.ebay.com/itm/403534100457

ForrestErickson, 2022-08-16:
IIRC I purchased from an Amazon source perhaps two years ago, while they were pictured assembled the I2C module came separately.   I have a 51% confidence that I soldered the I2C on to the display.

ForrestErickson, 2022-08-16:
It might be that you can order with and with out.

With:
[image]

?with out?
[image]

You can select just the I2C board for ordering alone too:
[image]

ForrestErickson, 2022-10-25:
Lawrence reports we have test from two vendors each of which had labeling on the PCB of "**2004A**"
Lawrence reports that both Mouser and DigiKey have equivalent parts.

Lawrence will change the merchant link in KiCAD Schematic (parts).

ForrestErickson, 2023-01-03:
@LokiMetaSmith  What is the status of this?