PubInv/general-purpose-alarm-device #57: I2C Addres of LCD set for A2, A1, and A0 low.

ForrestErickson, 2022-08-16:
Just FYI.

Referring to the schematic page 4 of 6, the typical default I2C peripheral address of LCD is set for the PCF8574 with A2, A1, and A0 high.  This would correspond to resistors on our aboard R305, R307 and R309 not stuffed (aka DNI).  However in the schematic I put in a value of 1Ohm so when built to our parts list  in the August 2022 build event, the address will be a different address than the default likely to be found in example code all over the internet.  The exact address can be figured out from the PCF8574 data sheet.

Their is no hardware reason to prefer one address over another.

This specific hardware configuration simply needs be noted for firmware development.

ForrestErickson, 2022-08-16:
Data sheet to be found at: https://www.nxp.com/docs/en/data-sheet/PCF8574_PCF8574A.pdf

Page 5 explains the address decoding.
[image]

Our schematic (BOM) calls out the PCF8574A.
So, If I read this correctly, the slave address will be 0x70 to write to the display.

ForrestErickson, 2022-08-17:
A good article, reference, for programing LCD features found here: 
https://lastminuteengineers.com/i2c-lcd-arduino-tutorial/

Note that customer characters can be created.
What custom characters might be of use for PolyVent?

Consider graphical like combinations of multiple characters too.

LokiMetaSmith, 2022-09-07:
i2c 0x38 has been tested to work with v1 of the board

LokiMetaSmith, 2022-09-13:
Change has been made to the GPAD test firmware

closed 2022-09-13 by LokiMetaSmith