PubInv/general-purpose-alarm-device #58: Create GPAD Factory Test Firmware

ForrestErickson, 2022-08-17:
Factory test requires firmware that exercises the features to allow for quick determination of feature functions.

ForrestErickson, 2022-08-17:
Lee has committed firmware: https://github.com/PubInv/general-alarm-device/tree/main/Firmware/GPAD_Factory_Test
It was simulated on Wokwi at:  https://wokwi.com/projects/340248560782017108

Press the button to light the five LEDs and make tone.
[image]

ForrestErickson, 2022-08-17:
As of 20220817 1200, this test does not cover the SPI Peripheral interface.
This test assumes the Arduino UNO boot loader is installed by a method TBD.

ForrestErickson, 2022-08-23:
As of 20220817 1200, this test does not cover independently test the LIGHT0-LIGHT4. A "walking one" test would be better.

LokiMetaSmith, 2022-08-23:
Task List:
- [ ] Add a function that will light in succession LIGHT 0 through LIGHT 4
- [ ] Add display text to describe the current test action and expected result
- [ ] Add a test function to verify SPI functionality
- [ ] Add a test function to verify i2c  functionality
- [ ] For i2c test for the ACK bit when testing the display function
- [ ] Exercise the display by showing all 20x4 characters as black squares
- [ ] Exercise the character table by walking through each one

ForrestErickson, 2022-09-13:
Note the Wokwi documentation located at:
https://docs.wokwi.com/parts/wokwi-lcd1602

indicated we can set an I2C address attribute for the simulation.

[image]

 The simulation should be updated for the address Lawrence found to work.

ForrestErickson, 2022-09-13:
Here is code with the Lawrence change.
[image]

ForrestErickson, 2022-09-13:
Lee has updated the I2C address of the display in the simulation at: https://wokwi.com/projects/340248560782017108

It now works on address 0x38 as does the hardware.

[image]

ForrestErickson, 2022-09-22:
Updated factory test firmware 20220922 for the buzzer at 4KHz and it is much louder. Unpleasant too.

ForrestErickson, 2022-10-19:
Detail on: Add a test function to verify i2c functionality

Make a conditional test that the exact address for the LCD is found during the I2C scan and if not warn the tester with a Serial.println() message.

RobertLRead, 2022-11-01:
This code has been reviewed and works well. New features may be requested with issues.

closed 2022-11-01 by RobertLRead