PubInv/general-purpose-alarm-device #59: Links To Images of Display PCB Assembly Broken on Main Readme.md

ForrestErickson, 2022-08-20:
On 20220820 Lee found the main readme.md at

https://github.com/PubInv/general-alarm-device

has no images ofr the PCB Assembly front and rear.
[image]

I am using my Windows 7 lap top with FireFox browser.

ForrestErickson, 2022-08-20:
Lee thinks he found the images in a new location at: \general-alarm-device\Hardware\Documentation\PICS

ForrestErickson, 2022-08-20:
Looks like history shows images moved.  My local git log also shows history of a "conflict" with some of the files in the hardware folder and Lee is not sure of the significance of this log message. 
I assume no action is needed it is just history of a conflict which is now resolved.

[image]

LokiMetaSmith, 2022-08-21:
Fixed this when I updated the latest enclosure rendering

closed 2022-08-21 by LokiMetaSmith