PubInv/general-purpose-alarm-device #60: PCB doesn't include a way to upload the Arduino Bootloader through the ISP

LokiMetaSmith, 2022-08-22:
[image]

In a standard Arduino Uno, there is an ISP header which is used to push the initial bootloader onto the device, so subsequent uploads can be pushed through the USB serial interface. 

This wasn't done with the GPAD. 

A work around is to use the pins brought out on the Arduino Uno header/hat.

ForrestErickson, 2022-08-23:
Re programing the boot loader.
Programing the first one will require some flying wires to a solder less bread board from an already programed UNO.
To the target GPAD connections can  be made for all but the /RESET signal on the RJ12 connector.  One more connection is required, a wire from the programing UNO at D10 to the target at the /RESET pin.

In hind sight a test point on the /RESET net would have been helpful for this operation.

Once we have one GPAD loaded with a boot loader it can program the others with much simpler connection.
To load the boot loader we need connections to the programing RJ12 to target RJ12 with the RJ12 DATA cable.  Connect from the programing GPAD pin D10 to the target GPAD at /RESET.
The RJ12 connection provides power, ground and the SPI interface.
We should document this in the Github repository so that persons building GPAD will not have to figure it out.  If you can make photographs of your setup that would be useful for such documentation.

A helpful note on programing the boot loader here: https://sparkfuneducation.com/how-to/installing-an-arduino-bootloader.html
[image]

Hope this helps.
Lee

LokiMetaSmith, 2022-08-30:
This worked, and will be applied to ver1 boards.

we need either a picture or SVG render of the GPAD plus a SVG render of an arduino with the wiring drawings shown to finish documenting this.

ForrestErickson, 2022-12-14:
Closing this because we have uploaded the ?dumoluvo? bootloader with success as documented in the manufacturing notes.

closed 2022-12-14 by ForrestErickson