PubInv/general-purpose-alarm-device #64: Wrong footprint on pcb layout for U301

LokiMetaSmith, 2022-08-29:
The datasheet lists the package as a SO16: plastic small outline package; 16 leads; body width 7.5 mm

where as the footprint chosen was Package_SO:SOIC-16W_5.3x10.2mm_P1.27mm, and does not accommodate the package

LokiMetaSmith, 2022-08-30:
We weren't able to find a replacement part in stock, this will require a board revision spin to correct.

ForrestErickson, 2022-08-30:
As of 20220830, Lee has reviewed the data sheets for NXP and for Texas Instrument. Neither vendor makes this part in the SOIC-16W_5.3x10.2mm_P1.27mm package. 

Working from memory, The KiCad library may have offered Lee the impossible  5.3mm package width. Will have to review. 

**Bottom line, with out a version 2 this PCB  will not be good for anyone else to use.**

RobertLRead, 2022-10-11:
Does that mean we can build our with a bit of rework, but that it would be
too much rework for someone else? So long as that is documented, I don't
view this as a problem; it is about what I would expect for a first try.

ForrestErickson, 2022-10-11:
Robert, regarding, "Does that mean we can build our with a bit of rework,"
Yes we can and have built.
I documented this and other parts of the manufacturing here:
https://github.com/PubInv/general-alarm-device/blob/main/Hardware/Documentation/ManufacturingUnitTestTroubleshooting.md

Screen shot here:
[image]

LokiMetaSmith, 2022-11-22:
confirmed resolved

ForrestErickson, 2023-03-01:
Version 2 PCB screen shot.
[image]
JLCPCB reports the part in our BOM is short stock
<html>
<body>

PCF8574AT | Package_SO:SOIC-16W_7.5x10.3mm_P1.27mm | PCF8574AT/3,512C83193

</body>
</html>

There is a similar NXP part in stock:
https://jlcpcb.com/partdetail/NxpSemicon-PCF8574AT_3518/C86832

PCF8574AT/3,518
Manufacturer NXP Semicon
MFR.Part # PCF8574AT/3,518
JLCPCB Part # C86832
Package SOIC-16-300mil
Description 100kHz 8 POR I²C SOIC-16_300mil I/O Expanders ROHS
Here is mechanical drawing from NXP data sheet:
[image]

This part will fit.
The I2C address is the same as the part we are using in Version 1.
[image]

Lee is going to put this JLCPCB Part # C86832 into the KiCAD documentation.

ForrestErickson, 2023-03-01:
20230301 Schematic library updated. Schematic updated. PCB updated for  JLCPCB Part # C86832.

RobertLRead, 2023-06-20:
Fixed in Version 2.