PubInv/general-purpose-alarm-device #76: Identify ESD Testing Requirements for GPAD as a Medical Product.

ForrestErickson, 2022-09-20:
What ESD testing requirements may their be for the GPAD as an alarm and display on a medical product?
To what UL or EN should we test and conform?

RobertLRead, 2022-09-21:
good question. I really have no idea.

ForrestErickson, 2022-12-14:
@Robert,

There is an EMC requirement among these:

FYI
Note while reading about buzzers I found a louder buzzer at DigiKey which references a medical requirement.
The part: https://www.digikey.com/en/products/detail/mallory-sonalert-products-inc/SBT5LMMVPC/5418645
Part Data sheet: https://www.mspindy.com/spec-sheets/SBT5LMMVPC.pdf references IEC 60601.

https://en.wikipedia.org/wiki/IEC_60601

IEC 60601 is a series of [technical standards](https://en.wikipedia.org/wiki/Technical_standard) for the safety and essential performance of medical electrical equipment

IEC 60601-2-12 Medical electrical equipment - Part 2-12: Particular requirements for the basic safety and essential performance of critical care ventilators

ForrestErickson, 2022-12-14:
@RobertLRead Regarding the IEC 60601 testing.

The next step might be to contact a "CB" aka Certified Body aka testing house such as Edison Testing Labs or UL and ask them about quoting such testing. We should be fare and tell them we are getting quotes and do not have near term plans for actual testing and perhaps Public Invention would want to consider a fund raising drive for such testing as a special case for near project completion.

RobertLRead, 2023-01-03:
Note this sub-field is specific to PolyVent:

IEC 60601-2-12 Medical electrical equipment - Part 2-12: Particular requirements for the basic safety and essential performance of critical care ventilators

RobertLRead, 2023-01-03:
Lee and Rob agree this should remain open but be a low-priority for now.