PubInv/general-purpose-alarm-device #79: J101 footprint mechanical size

LokiMetaSmith, 2022-10-01:
Tolerance on the mechanical retention holes on the footprint are too large and leaves a lot of room for movement.

ForrestErickson, 2022-11-13:
From drawing here are footprint dimensions.  

[image]

Hole size in PCB one of two.
[image]

Update hole size two of two.
[image]

ForrestErickson, 2022-11-13:
Save changes into the GPAD library and overwrite.
[image]

LokiMetaSmith, 2022-11-22:
Confirmed Resolved

closed 2022-11-22 by LokiMetaSmith