PubInv/general-purpose-alarm-device #81: Capture Bow on Version 1 PCB, 0.008". Speculate on Tweeking Out on Serial Numbers 7-15.

ForrestErickson, 2022-10-04:
For the Version 1 PCB from JLCPCB, Lawrence found a slight bowing to the PWA.
Lee found for a raw PCB with a straight edge and feeler gauge that that 0.010 would not fit but 0.008" would.

[image]

Lee would think this is natural given the unequal copper on the top and bottom of the PCB and may be entirely batch to batch dependent and may be negligible.

However, for Version 1, PWAs, Serial Numbers 7-15, we might try to hold flat the GPAD PCB when soldering the LCD module at the 16 pin connector to

LokiMetaSmith, 2022-10-25:
This issue can be closed when added to the documentaion

RobertLRead, 2022-11-01:
We agree.

closed 2022-11-01 by RobertLRead