PubInv/general-purpose-alarm-device #85: Can't upload firmware

LokiMetaSmith, 2022-10-17:
After initially loading the firmware, it is trivial to upload firmware over the serial interface, but later firmware uploads seem to have issue with communicating with the board. 

It appears that even without  the DTR line, we can use the RTS line as it is toggled at the same time. https://forum.arduino.cc/t/does-avrdude-toggle-rts-and-dtr-for-ftdi-reset/635894/2

The suggested wire mod is a 100nF capacitor from pin 4 of U501 to  the nRESET net If that is insufficient, a 100 ohm resistor may also be used. 

In the reference Uno schematic, there is no filter capacitor on the reset line, one potential mod is to tombstone C110 such that it is connected to nRESET and pin 4 of U501.

ForrestErickson, 2022-10-17:
Here I document the problem. I change the version in the sketch to something I can distinguish such as 88 and try to upload the sketch WITH OUT pressing the reset button.
 
 The target was stuck in the "Uploading..." mode 
[image]

After about 60 seconds I gave up and pressed the reset button and the IDE displayed the error message like this.
[image]

ForrestErickson, 2022-10-17:
Regarding the timing on pressing the reset button when the "Uploading..." message status displays.

I find that If I press as soon as I see the message I get success every time. 
[image]

However, if I count to about two seconds I usually get a failure.
[image]

LokiMetaSmith, 2022-10-18:
confirmed the workaround, still needs a hardware fix.

LokiMetaSmith, 2022-10-25:
Attempted to rework the U501 but the pitch on the device is too small to reliably fix. 

Moving this to HW change to V2 of the pcb

ForrestErickson, 2022-11-12:
# 20221112 Lee tried the rework.  
First confirm the DUT was working normaly. It enumerated as COM 16.

## Make modification step by step
Only shorted U501 pin 4 to pin 5. Connected the DUT to the PC. The DUT did not enumerate as a COM device.
[image]

Note that TP 203 is high.

### Remove the short from U501 pin 4 to pin 5.
The DUT now enumerates.
[image]
Note that TP 203 is low.
Therefor we cannot simply short from U501 pin 4 to pin 5 to get easy mechanical access to the nRTS pin.

### Move C601 and Wire to U501 Pin 4 Only
Second rework attempt 
[image]

Apply power. 
The DUT enumerates as normally.
[image]
The DUT resets when the USB is attached.

The DUT resets when the Arduino IDE opens the serial monitor as a normal UNO would.
[image]

## Conclusion 
Though very difficult, this rework works so that the GPAD programs like an UNO.  Pressing Reset button is not required.

@RobertLRead 
@LokiMetaSmith 
Please see the above notes.

ForrestErickson, 2022-11-12:
@RobertLRead 
@LokiMetaSmith 
Will either of you be able to attempt this rework?

I used ?30 AWG? wire wrap wire and lots of magnification to see what I was doing. 
I am not excited about it but am willing to rework any and all units especially units that firmware developers will have.
Once practiced, I am guessing the rework will take me all of about ten minuets each.

Please advise.

LokiMetaSmith, 2022-11-22:
Confirmed resolved, the units can be programmed as described above

closed 2022-11-22 by LokiMetaSmith