PubInv/general-purpose-alarm-device #86: update the diagram.json wokwi simulation for reworked gpad

LokiMetaSmith, 2022-10-18:
https://github.com/PubInv/general-alarm-device/blob/main/simulation/diagram.json

update simulation diagram from v1 to include reworks, SPI nCS to D10, LIGHT4 D7

ForrestErickson, 2022-10-19:
Lee has added a full copy of the simulation of the Factory Test at: https://github.com/PubInv/general-alarm-device/tree/main/simulation/FactoryTest
LIGHT4 is on D7
This simulation now has a logic analyzer to monitor the SPI bus and nCS is on D10.
Lee also added the I2C bus to the logic analyzer inputs.  There is a screen shot there.

Is it necessary to update the diagram at 'https://github.com/PubInv/general-alarm-device/blob/main/simulation/diagram.json' as it does not seam to be associated with a particular simulation ?
Or if it is it should be moved into a folder with that simulation?

Could it be deleted?

Lee has added Robert to the assignees because I think we need to mutually plan on how to use these folders and files to cover the diversity of simulations we may want.

ForrestErickson, 2023-01-03:
@LokiMetaSmith 
Can you review the status.
I hope we can close this.