PubInv/general-purpose-alarm-device #87: Create PolyVent SPI card which makes direct connection go the GPAD SPI interface

RobertLRead, 2022-10-18:
One of the driving use cases for the GPAD is the [PolyVent](https://gitlab.com/polyvent) open source educational ventilator. All medical devices require some sort of alarm system.

This issue is the extension of the PolyVent with a new board (it is designed to be extended in this way) that provides a RJ12 cable connection to the GPAD so that the GPAD can be mounted on the case of the PolyVent to allow an alarm mechanism. This will also require the PolyVent to make modifications to [Project VentOS](https://gitlab.com/project-ventos/ventos) which is the software which runs on the PolyVent, and is capable of raising alarms. Talk to Robert L. Read about this as a bridge between the projects.

ForrestErickson, 2022-10-19:
Related. 

If it does not exist, make a PCB card extender for PolyVent chassis. 

Here is something like what I have in mind.
[image]

ForrestErickson, 2022-10-27:
Lawrence made a template PCB and Lee has created schematic and PCB and added it to the repository at:
https://github.com/PubInv/general-alarm-device/tree/main/Hardware/PolyVent_SPI_Controller

ForrestErickson, 2022-10-27:
The next step will be for Lee to review the schematic and the PCB with Lawrence.

If Lawrence can check it out first that will be of help.

Lee plans to update the README with more information.

RobertLRead, 2022-11-01:
Rob comments: Public Invention is happy to order 5 of these when they are ready to support the PolyVent project. Maybe Lee or Lawrence can populate with components, and then ship a few cards to Rob (and separately to Nathaniel?).  This will allow us to really connect the GPAD to the PolyVent, which will be a nice addition to the PolyVent. This will also enhance our publication we submit to HardwareX if the two projects cooperate.

ForrestErickson, 2022-11-08:
During Review on 20221108 by Lawrence and Lee :

Suggest change R121, R22 and R123 from resistors to polyfuse.
Add power indicator LEDs for +24/12, +5, +3.3 power.
Add 7812 or a switcher model 
Make voltage output to I2C resistor selectable like for the RJ12.  This voltage can be selected independent of the voltage selected for RJ12.

Consider redraw schematic with out global power and ground to the sub sheet.

ForrestErickson, 2022-11-10:
@LokiMetaSmith 
Regarding, 

> It should be selectable, I would use a rotary switch, jumpers, dip switches, or resistors, because it can possibly have several different devices down stream, and modules can be combined or unnecessary over time. 

Sorry, I had a different point to my question.
I have not figured out how to look at the PolyVent in any meaningful way including schematics.
I have no idea if any of the possible nCS lines I chose are appropriate.
Do you have any idea or way to check that any of these lines can be used as nCS with out conflict?

Lee

ForrestErickson, 2022-11-23:
20221123 Update schematic for:

Suggest change R121, R22 and R123 from resistors to polyfuse.
Add power indicator LEDs for +24/12, +5, +3.3 power.
Make voltage output to I2C resistor selectable like for the RJ12. This voltage can be selected independent of the voltage selected for RJ12.

CAUTIUON. Components have been reannotated so the reference designators above have possibly changed.

ForrestErickson, 2022-11-24:
I have updated the Schematic and PCB with these changes and made a PDF of schematic. Ready for review.

ForrestErickson, 2022-11-24:
Note: The schematic symbol and PCB footprint I used for the buck converter I found at:
https://forum.kicad.info/t/lm2596-step-down-module-dc-dc-buck-converter-power-supply/23904/10?u=forresterickson

ForrestErickson, 2022-11-26:
20221126 Lee created Gerbers and drill files and zipped them up. Created BOM and Parts list with parts populated for level shifter towards possible JLCPCB build.  Added 3D footprint for buck regulator.

ForrestErickson, 2023-01-03:
Closing as done per Robert's success.

closed 2023-01-03 by ForrestErickson