PubInv/general-purpose-alarm-device #88: Fully expose LCD, LED, and Audio functions in a "full robotic API"

RobertLRead, 2022-10-18:
The GPAD firmware needs to support two APIs (conceptually, they might be part of one.)

An abstract module defines "alarms" at 5 levels with a text description. This is designed to be simple to use by an external device or tool, and to be a standard that will survive hardware changes to the GPAD.

However, it it also use to fully expose ALL of the features of the GPAD if a programmer wants to make a more detalied or general use of the GPAD. For example, the GPAD could implement a very simple game.

Ideally, this API will be abstract in the sense that the SAME API is exposed by SPI and the serial port. Both of these might be exposed by a remote procedure call interface (that is, a C library). But a byte-level protocol is also useful, and a human-typable protocol is also useful from the serial port for testing. The goal of this issue is fully define and implement this API, with at least one binding where it can be tested.

ForrestErickson, 2022-10-18:
Note: I have used the depreciated terms Master and Slave in the above note about the I2C bus only in this first explination of the hardware capability **ONLY** for consistency with the data sheet for the ATmega328.   

We should migrate to the terms "Controller" and "Peripheral" (Just like for the new terms for SPI) for the I2C bus in our code and documentation.

ForrestErickson, 2022-10-18:
The mute button function should be exposed.  

Here is a possible wrinkle, When the mute button is pressed an alarm is muted for a time TBD (Perhaps programable). Perhaps a function where the controller (Poly vent) poles the button status could return the time left for MUTE from a variable.  Zero mS is returned when unmuted.

Anther function might return the actual state of the button at the time of the poling.

There is a library Lee found useful for debouncing a button and detecting long presses. Look for:  DailyStruggleButton at
https://github.com/cygig/DailyStruggleButton