PubInv/general-purpose-alarm-device #89: Improve the code if the I2C scan for the Liquid Crystal is not found

RobertLRead, 2022-10-19:
This is part of this issue:
[Issue #58 ](https://github.com/PubInv/general-alarm-device/issues/58)

We believe a good first issue would be to improve the output of on the Serial Port if the I2C Liquid Crystal Display is not found, which would probably be a hardware error.

ForrestErickson, 2023-01-03:
@NULLbodyHERE 
Hello and happy new year!

Have you seen this issue?
Will you have interest and time to work on this?

NULLbodyHERE, 2023-01-04:
@ForrestErickson 
Happy New Year to you too!

Yes, absolutely! :) It was on my to-do list since my last PR and I thought I'd wait to receive the GPAD prototype to make (hopefully) better contributions. I shall get properly stuck into it soon after I recover from this lovely, perfectly timed chest infection. I will shortly get back to you on this matter. Thanks.

RobertLRead, 2023-01-31:
Dear @NULLbodyHERE ,

I hope you are recovered now.  If you have time, we would like you to undertake this issue. Can you get back to us on whether you can undertake this or not?

Lee would be glad to find a convenient time via video chat to confer with you on this.

NULLbodyHERE, 2023-02-02:
Hello @RobertLRead,

Thanks. I should be able to get around to it this weekend. I had hoped that maybe we could sort out my GPAD screen issue so I could use the physical prototype to write better code. I'm happy to proceed nonetheless, using the simulator. I will discuss more with @ForrestErickson as to when we can arrange a video call. Thanks for your patience thus far.

ForrestErickson, 2023-02-03:
Hello Seyi  @NULLbodyHERE ,
I live in the Eastern Time zone which is currently UTC-5.  
My Monday and Wednesday mornings and early afternoon's are tied up with a class for some months. But with some advanced notice other days and times are available.

@RobertLRead 
Is there a way for us to use the video conferenceing system at:
https://publicinvention.frameable.com/workspace-3983651163842326/1176682421602418?t=957195986196474
I assume that it is available 24-7. and that Seyi may have access.  **Can you confirm this is how we might work?**

NULLbodyHERE, 2023-02-11:
Hello @RobertLRead,

I have made significant improvements on the I2C scan such that the program now stably handles timeout errors. In the I2C scanning mode, the scan is in a continuous looping state, where the terminal is updated once every second so as to indicate that: 

1. The scan is still in process and;
2. There are found addresses.  

Upon pulling SCL or SDA to ground, the scan is paused and an immediate error message is thrown. The program then attempts to request data and read from the address in question, printing to the terminal the results. Then there is a breakdown message of total errors showing the number of addresses attached to a specific error type. For now, this is only applicable to the target address since every other address is unconfigured and would clog the terminal unnecessarily.

Since we know that we are looking for a specific address and a specific number of devices, the terminal will also notify you whether:

1. The target number of addresses have been found, and;
2. The number of addresses found (including none) despite not reaching the target.

A prompt message will then require the pushing of the mute button to resume the scan. The program should no longer freeze (hopefully), and selection of other test groups can continue. However, there is probably still room for improvement to make it completely robust.

@ForrestErickson please let me know the results of your tests so I can improve on the code. I have tested it several times on my GPAD (SN 1) prototype (just using the terminal with external devices) and made careful adjustments to the I2C timeout variable through experimentation. Feel free to modify the macro `WIRE_TIMEOUT` if you discover a consistent failure to find any address. Also, the Wokwi simulator doesn't seem to take kindly to the 'improvements'. When you break a wire and play the simulation, no messages are printed to the terminal except `--- BEGIN I2C SCAN ---` and `--- TERMINATE I2C SCAN ---`. It is quite limited after all.

Next, to refactor the other sections from the #144 improvement.

Cheers!

RobertLRead, 2023-02-11:
Dear Seyi,

This sound pretty cool. I'm a little hazy on what the original problem was,
but it sound like you're doing a great job addressing it.

ForrestErickson, 2023-02-12:
@NULLbodyHERE 

SWEET!

I will add to my growing back log to test this too.

Regarding, " When you break a wire and play the simulation, no messages are printed to the terminal except --- BEGIN I2C SCAN --- and --- TERMINATE I2C SCAN ---. It is quite limited after all."

Could you try a switch to ground as a way to impair the I2C bus during run time?

NULLbodyHERE, 2023-02-14:
@ForrestErickson 

As far as I'm aware, you can't dynamically change the state of the simulation. You must stop it, make amendments, and then play it. Even when I do stop it and change either the SDA or SCL to ground or just break the line, It still produces only the messages "BEGIN" and "TERMINATE" for some reason.  Please correct me if I'm wrong about the simulation. In any case, it will require further investigation. 
Cheers.

ForrestErickson, 2023-02-14:
Send me a link to your simulation please.Forrest Lee EricksonCell: 865-518-1040Sent from cell.

  for the Liquid Crystal is not found (Issue #89)

NULLbodyHERE, 2023-02-16:
@ForrestErickson 
It's this link: https://wokwi.com/projects/345960281987351124 except that you need to manually copy and paste the updated code, since the page hasn't been updated to the latest firmware version.

ForrestErickson, 2023-02-16:
@NULLbodyHERE ,
Oh I inferred that you had set up your own Wokwi simulation. This is a link to mine. Also I had assumed you had experimented with additional code changes not yet committed.

Please clarify.

I have made a Wokiwi copy with the V1.32 code and added buttons to short the SCL and SDA lines during run time.
Try it : https://wokwi.com/projects/356858915742940161

_I sure am having fun!_

ForrestErickson, 2023-02-16:
@NULLbodyHERE 

Here I made a run time error by pressing the Yellow button during the LCD test.
[image]

NULLbodyHERE, 2023-02-18:
@ForrestErickson 
In future, I'll make sure to set up my own link to reference.  

> Also I had assumed you had experimented with additional code changes not yet committed.
No, this was with code that had been committed in v1.31.

In any case, it looks like you have creatively come up with a workaround to test that part of the code virtually. Great idea! 
Cheers.

ForrestErickson, 2023-02-18:
@NULLbodyHERE 
Regarding, "In any case, it looks like you have creatively come up with a workaround to test that part of the code virtually."
Yup it was a delight to do and wanted to share with you.

I am about to start my day which will be to take a telescope to our local library with a solar filter so I am about to be unavailable. Hope your day was a good one.

RobertLRead, 2023-06-20:
fixed.

closed 2023-06-20 by RobertLRead