PubInv/general-purpose-alarm-device #91: Refactor the test functions in the Factory Test Code...

RobertLRead, 2022-10-19:
At present, you can test the GPAD with [the code installed "at the Factory"](https://github.com/PubInv/general-alarm-device/tree/main/Firmware/GPAD_Factory_Test) (that we install on each unit), by pushing a button.

However, this code combines a test of the buzzer and the LEDs, and doesn't test everything. A better method would be to have separate test functions for each operational capacity we want to test, and to cycle through those test by repeatedly pressing the mute button. 

(We can also test by writing a simple parser of the serial port, but that is too complicated for the most basic hardware test capability.) 

This issue is to refactor that code effectively.

This may require "debouncing" the mute button, to correctly count the number of times it has been pressed. Lee has suggested this code: [https://github.com/cygig/DailyStruggleButton](https://github.com/cygig/DailyStruggleButton) may be a good way to do that.

NULLbodyHERE, 2022-11-12:
Hi @RobertLRead,

I'm currently tackling this issue and have a few questions...

Do you want the setup() code modified such that peripherals are only initialised, but nothing is tested (e.g. flashing LCD) until the mute button toggles through each one? I ask because perhaps you would like to further test the functionality of an operation even after it has displayed its first signs of working properly.

Also, so far, I have:
- testSwitchMute()
- testLEDs()
- testBuzzer()
- testI2C()
- testLCD()

Does this suffice as every operational capacity you would like to test, or should this be modified?

ForrestErickson, 2022-11-12:
Regarding, "...by pushing a button."  The button in question is labeled MUTE and has reference designator S601.
See in this image: https://user-images.githubusercontent.com/5296671/195462489-233d7f4f-62ae-4d29-930c-db42419f31b2.JPG

ForrestErickson, 2022-11-12:
Hello @NULLbodyHERE 
FYI, we have a simulated simplified GPAD at this Wokwi location.
https://wokwi.com/projects/345960281987351124
See: 
[image]

You could clone your own copy for testing. Then you could "break" the simulated hardware on your Wokwi by changing the circuit. For example go into the Wokwi tab diagram.json and change the address of the I2C display to something other than the expected value (highlighted in this screen shot)
[image]

This would allow you test code that in the factory test reports the all the I2C devices found and further identifies that the expected address was NOT found.   

CC: @RobertLRead

ForrestErickson, 2022-11-12:
Hello @NULLbodyHERE 
There is a schematic for the GPAD at this link: https://github.com/PubInv/general-alarm-device/blob/main/Hardware/GeneralPurposeAlarmDevicePCB/PDF/Schematic-GeneralPurposeAlarmDevicePCB-V120220810.pdf

**CAUTION.** This "Rev 0.0" schematic does not reflect a rework interchanging D7 and D10.  
When I get back to working on updating hardware documentation this will change and I will update the schematic version and add some descriptive note too.

CC: @RobertLRead

ForrestErickson, 2022-11-12:
Hello @NULLbodyHERE 

Regarding, 

> Do you want the setup() code modified such that peripherals are only initialized, but nothing is tested (e.g. flashing LCD) until the mute button toggles through each one? I ask because perhaps you would like to further test the functionality of an operation even after it has displayed its first signs of working properly.

### Short Answer 
Test it all at setup() and report any errors that firmware alone can detect and then in the loop() allow the user to test again one at a time.

### Detailed Answer
I had the role of hardware designer and wrote the initial "[GPAD_Factory_Test.ino](https://github.com/PubInv/general-alarm-device/blob/main/Firmware/GPAD_Factory_Test/GPAD_Factory_Test.ino)"
I deliberately left some tests for others to write as a way others could contribute to the project as you are doing.  THANKS!

However from a hardware testing role some things I want. The setup() sets the LED on pin D13, the "LED_BUILTIN" high and on near the start of setup() and low again at the end of setup(). This is so that I can see with my eyes and with an oscilloscope the setup() function has run which informs me that firmware is loaded and running and that lots of the components around U102 the Atmega328 are correct.   So that test remains in the setup().  

I wrote the "splash" messages to the LCD and to the serial port and put them in the setup() so that I could tell when the firmware ran that I had uploaded successfully and that for a device under test that was normal a lot of the hardware was correct. What the GPAD_Test.ino did not do well was help trouble shoot a mal function once one was identified.  
If for example pins shorted on the I2C buss expander U301 to the nibble wide input to the LCD U302 some characters would not display.  Here is schematic detail:
[image]
If U301 P4 and P5 were shorted together then only when both were set low or both set high would the hardware display the intended character.  Do you follow?  So for a person troubleshooting it would be helpful to have independant tests in the main loop() which they could select and run and rerun as they trouble shoot.
I hope this helps you anticipate the answer to " I ask because perhaps you would like to further test the functionality of an operation even after it has displayed its first signs of working properly."  The answer is yes.

### More on Factory Hardware Testing Mind Set
Factory Hardware Testing is typical looking for three common errors
1. Missing components
2. Solder shorts where tracks and pins are close
3. Intermittent connections. The most horrible of cases for the person troubleshooting.

The tests in setup() may find most of number 1.
It takes some ingenuity to figure out the tests for number 2 such as lighting the LIGHT1 through LIGHT4 sequentially one at at time AND lighting all but one sequentially one at a time.
For number three you need a test that runs with dynamically changing data continuously so that the user can poke and prod at the device under test. Tapping with fingers or even rubber mallets may get used. There is no end in sight to the things you might device and at some point you just stop and wait till a person troubleshooting comes to you and says. I need help.

### Notes
Regarding the function "testLEDs()"  It should be renamed to reflect signal names on the hardware which are "Light0", "Light1",... "Light4".   Though in the code perhaps they should be upper case and I shail make an issue to rename these nets in our version 2 PCB.

While writing this note to you I noticed that page 6 which has the circuits is missing from the schematic above. 
Here is a screen shot of the missing page.
[image]

I hope the way I have provided information is not too tedious and helps you with insight that I in the role of someone troubleshooting for manufacturing errors on a particular device would want and need from factory test firmware.  I hope it also helps you mentally assume that role so that you can help others with anticipated solutions for factory testing other products.

Thanks so much for your involvement and questions.  It helps.

CC: @RobertLRead

NULLbodyHERE, 2022-11-13:
Hello @ForrestErickson,

Thank you very much for taking the time to reply and provide me with a such a comprehensive response! It was just about the right information to clarify, and luckily, I have some hardware design experience myself, so the schematics aren't too scary. I shall take everything mentioned into consideration as I continue to expand code functionality.

Fortunately thus far, I have exclusively been using your Wokwi GPAD simulation to verify the code and be certain of its functionality; it has proven to be invaluable. I have already managed to use the mute button (paired with the DailyStruggleButton debouncing library) to toggle through tests for the button itself, the LEDs, and the buzzer.

Regarding the renaming of the function `testLEDs()`, I should have mentioned that my approach was to group the tests such that you can toggle through each group with a double press, but through each individual test with a single press. This facilitates readability for potential further expansion of tests. So, `testLEDs()` is an umbrella function that currently encapsulates the testing of both the built-in LED _and_ the external LEDs (or 'Lights'). Would you still prefer I rename the function? Or perhaps, you can modify my PR before merging. I have also created a feature to automatically toggle between all tests within a particular test group that is both enabled and disabled at the command of a long press.

Looking forward to completing this challenge! Feel free to assign me this task if you wish. Thanks again for the response.

CC: @RobertLRead

ForrestErickson, 2022-11-13:
Hello @NULLbodyHERE 
Regarding, "Would you still prefer I rename the function?"
Not necessary. 
Please put descriptive comments into the code such as: 

> testLEDs() is an umbrella function that currently encapsulates the testing of both the built-in LED and the external LEDs (or 'Lights').

And thanks again.

ForrestErickson, 2023-01-03:
Lee to refresh his memory of the status of this.