PubInv/general-purpose-alarm-device #93: Change the use of PIN 13 for wink LED to reserve it for SPI...

RobertLRead, 2022-10-25:
PIN 13 on an UNO is being used for the builtin LED. This is needed by SPI.

We need to make PIN 13 avaiable. The question of maintaining a visual heartbeat is yet to be resolved. Possible we can add an LED elsewhere.

ForrestErickson, 2022-11-13:
In version 2 PCB, the WINK LED has beenadded to ATmega PC3 pin which is A3 in Arduino UNO world.

ForrestErickson, 2022-11-13:
@RobertLRead 
@LokiMetaSmith 

For Version 2 PCB, I have added a new LED which can be used for the WINK_LED. See above.

LokiMetaSmith, 2022-11-22:
Confirmed Resolved

ForrestErickson, 2022-11-22:
I have reopened this and think it should only be closed by @RobertLRead when he has changed the firmware to reflect this.
My intent on this issue was to alert that the firmware is also impacted.

Version 1 PCB of course does not have the new LED so the result of change to firm ware is we will no longer have a winking LED to show us the status that the main loop is running.  If the users were to add a resistor and LED to the A3 pin on the UNO shield pins then the user would have the new functionality.

Screen shot showing how the tentative Version 2 PCB will have the new LED and resistor.
[image]

ForrestErickson, 2022-11-22:
@RobertLRead 

Do you have some LEDs and something like 330 Ohm resistors with which to rework your PCBs?
Do you have the desire to do so?

Please advise,
Lee

ForrestErickson, 2022-12-20:
@RobertLRead 

Note above: "I have reopened this and think it should only be closed by @RobertLRead when he has changed the firmware to reflect this.
My intent on this issue was to alert that the firmware is also impacted."

Do we close this out and I make a separate issue for Firmware change or do we close this when firmware is changed?

I see above that this issue is marked Firmware. (I am guessing I marked it so.)

Please advise.